ros-navigation/navigation2 #371: DWB controller fails to receive plan

crdelsey, 2018-11-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source, debug build of everything.
- Version or commit hash:
  - latest of everything as of Nov 27, 2018

#### Steps to reproduce issue
Start up nav stack using launch script

Send nav goal using RVIZ2.
#### Expected behavior

DWB controller starts executing

#### Actual behavior

Simple navigator says it sent plan to controller but controller never says it received it.

```
[INFO] [simple_navigator]: Received path of size 175 from planner
[INFO] [simple_navigator]: Sending path to the controller to execute.
```
There is no message from DwbController saying

```
RCLCPP_INFO(get_logger(), "Starting controller");
```

#### Additional Info

A quick look at the threads in DwbController didn't show any of them blocked in a bad way. It was like the message was never received.

mkhansen-intel, 2018-11-28:
Were you getting past the "CostMap 2D waiting for transform" loop for the local costmap?

mkhansen-intel, 2018-11-28:
I see this same problem, here's my log dump:
```
[amcl]: Setting pose (16.025000): -2.000 -0.500 0.000
[INFO] [simple_navigator]: Begin navigating to (0.00, 2.00)
[INFO] [navfn_planner]: Attempting to a find path from (0.03, 0.00) to (0.00, 2.00).
[INFO] [world_model]: Received costmap request
[INFO] [navfn_planner]: Making plan from (0.03,0.00) to (0.00,2.00)
[INFO] [navfn_planner]: Found valid path of size 1
[INFO] [navfn_planner]: Publishing the valid path.
[INFO] [navfn_planner]: Successfully navigated to (0.00, 2.00) with tolerance 2.00
[INFO] [simple_navigator]: Achieved navigation goal of (0.00, 2.00)
[INFO] [simple_navigator]: Received path of size 1 from planner
[INFO] [simple_navigator]: Sending path to the controller to execute.
[INFO] [FollowPathNode]: Starting controller
[WARN] [pluginlib.ClassLoader]: given plugin name 'lib/libstandard_traj_generator' should be '/libstandard_traj_generator' for better portability
[WARN] [pluginlib.ClassLoader]: given plugin name 'lib/libsimple_goal_checker' should be '/libsimple_goal_checker' for better portability
[INFO] [FollowPathNode]: DWBLocalPlanner
[WARN] [pluginlib.ClassLoader]: given plugin name 'lib/libdwb_critics' should be '/libdwb_critics' for better portability
[INFO] [FollowPathNode]: Using critic "RotateToGoal" (dwb_critics::RotateToGoalCritic)
[INFO] [FollowPathNode]: Using critic "Oscillation" (dwb_critics::OscillationCritic)
[INFO] [FollowPathNode]: Using critic "ObstacleFootprint" (dwb_critics::ObstacleFootprintCritic)
[INFO] [FollowPathNode]: Using critic "GoalAlign" (dwb_critics::GoalAlignCritic)
[INFO] [FollowPathNode]: Using critic "PathAlign" (dwb_critics::PathAlignCritic)
[INFO] [FollowPathNode]: Using critic "PathDist" (dwb_critics::PathDistCritic)
[INFO] [FollowPathNode]: Using critic "GoalDist" (dwb_critics::GoalDistCritic)
[INFO] [FollowPathNode]: Initialized
[INFO] [simple_navigator]: Control task completed.
[INFO] [amcl]: AmclNode publishing pose
```

crdelsey, 2018-11-28:
No everything was up and running at this point.

I only saw this problem once, so it may not be reproducible. I just wanted to capture the various issues I was running into so I could focus on one at a time without losing the others.

This one probably isn't a high given it is pretty rare.