ros-navigation/navigation2 #375: NavFn returns path of size 1

mkhansen-intel, 2018-11-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue
Follow nav2_bringup instructions

#### Expected behavior
Given the goal pose, the NavFn planner should return a valid path or fail.

#### Actual behavior
It returns a path of size 1, which can't be executed by the DWB controller.

```
[INFO] [simple_navigator]: Begin navigating to (0.00, 2.00)
[INFO] [navfn_planner]: Attempting to a find path from (0.02, 0.01) to (0.00, 2.00).
Detected time source change. Clearing TF buffer.[INFO] [world_model]: Received costmap request
[INFO] [navfn_planner]: Making plan from (0.02,0.01) to (0.00,2.00)
[INFO] [navfn_planner]: Found valid path of size 1
[INFO] [navfn_planner]: Publishing the valid path.
[INFO] [navfn_planner]: Successfully navigated to (0.00, 2.00) with tolerance 2.00
[INFO] [simple_navigator]: Achieved navigation goal of (0.00, 2.00)
[INFO] [simple_navigator]: Received path of size 1 from planner
[INFO] [simple_navigator]: Sending path to the controller to execute.

```
```
[INFO] [FollowPathNode]: Initialized
[ERROR] [DWBLocalPlanner]: No valid trajectories out of 399! 
[ERROR] [DWBLocalPlanner]: 0.01: ObstacleFootprint/Trajectory Hits Obstacle.
[ERROR] [DWBLocalPlanner]: 0.99:   PathDist/Trajectory Hits Unreachable Area.
[INFO] [FollowPathNode]: No valid trajectories out of 399! 
[ERROR] [simple_navigator]: Control task failed.
```

mkhansen-intel, 2018-11-30:
@orduno - any idea why this happens?

mkhansen-intel, 2019-02-19:
We haven't seen this in months, so I'm downgrading to Medium, we'll see if this occurs again in system testing.

crdelsey, 2019-07-01:
No longer a problem

closed 2019-07-01 by crdelsey