ros-navigation/navigation2 #390: Ros2 navigation cannot work in simulation(Gazebo).

congliu0913, 2018-11-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - initial_ros_setup.sh
- Version or commit hash:
  - nav2_bringup_readme branch
- DDS implementation:
  - default
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
  As we follow https://github.com/ros-planning/navigation2/tree/nav2_bringup_readme/nav2_bringup to launch Gazebo and load map by ros2 navigation, the simulation no response while send Goal by “2D Nav Goal” or set initial pose by “2D Pose Estimate” 

Terminal 1:
```
$ export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:~/turtlebot3_ws/src/turtlebot3_simulations/turtlebot3_gazebo/models
$ cd ~/turtlebot3_ws/src/turtlebot3_simulations/turtlebot3_gazebo/worlds
$ gazebo --verbose turtlebot3_ros2_demo.world -s libgazebo_ros_init.so
```
```
Terminal 2:
$ ros2 launch turtlebot3_bringup turtlebot3_robot.launch.py
```
```
Terminal 3:
$ ros2 param set /robot_state_publisher use_sim_time True
```
```
Terminal 4:
$ ros2 launch nav2_bringup nav2_bringup_1st_launch.py map:=<full/path/to/map.yaml> use_sim_time:=True
```
Download map from https://github.com/ROBOTIS-GIT/turtlebot3/tree/master/turtlebot3_navigation/maps
```
Terminal 5:
$ ros2 run rviz2 rviz2
```
Now we can show map in rviz and same as gazebo turtlebot3_ros2_demo.world
```
Terminal 6:
$ ros2 launch nav2_bringup nav2_bringup_2nd_launch.py
```
```
Terminal 7:
$ ros2 param set /world_model use_sim_time True
```

[image]

Now press “2D Nav Goal” button to send goal or press “2D Pose Estimate” button to set initial pose in RVIZ2.

#### Expected behavior
1. TB3 can get initial pose when send initial pose by "2D Pose Estimate" in rviz, and can show status in Gazebo
2. TB3 can go to set point when send goal by "2D Nav Goal" in rviz, and can show status in Gazebo

#### Actual behavior
TB3 no response in Gazebo what ever send goal or send initial pose.
#### Additional information
```
$ ros2 launch nav2_bringup nav2_bringup_1st_launch.py map:=/home/intel/Documents/map/map.yaml

[WARN] [amcl]: Laser scan has not been received (and thus no pose updates have been published). Verify that data is being published on the scan topic.
```
```
$ ros2 run rviz2 rviz2 

[ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /home/intel/navigation2/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[ERROR] [rviz2]: Failed to create 1 swatches_
[INFO] [rviz2]: Creating 2 swatches_
[ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /home/intel/navigation2/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[ERROR] [rviz2]: Failed to create 2 swatches_
[INFO] [rviz2]: Creating 4 swatches_
[INFO] [rviz2]: Setting goal: Frame:map, Position(-0.437653, 1.67005, 0), Orientation(0, 0, -0.73382, 0.679344) = Angle: -1.64786
[INFO] [rviz2]: Setting goal: Frame:map, Position(-0.580605, -1.68634, 0), Orientation(0, 0, -0.0331195, 0.999451) = Angle: -0.0662511
[INFO] [rviz2]: Setting goal: Frame:map, Position(-0.397456, -0.531156, 0), Orientation(0, 0, -0.765699, 0.643199) = Angle: -1.74425
[INFO] [rviz2]: Setting goal: Frame:map, Position(-0.438763, 1.36794, 0), Orientation(0, 0, -0.743401, 0.668846) = Angle: -1.67628
[INFO] [rviz2]: Setting goal: Frame:map, Position(-0.479174, -1.90133, 0), Orientation(0, 0, -0.742486, 0.669861) = Angle: -1.67355
[INFO] [rviz2]: Setting goal: Frame:map, Position(-0.393004, 1.66041, 0), Orientation(0, 0, -0.718478, 0.69555) = Angle: -1.60322
Parsing robot urdf xml string.
[INFO] [rviz2]: Setting goal: Frame:map, Position(-0.515518, 1.53056, 0), Orientation(0, 0, -0.688642, 0.725101) = Angle: -1.51923
```

```
$ ros2 launch nav2_bringup nav2_bringup_2nd_launch.py

Robot: Can't return current pose: Initial pose not yet received.
[WARN] [simple_navigator]: Current robot pose is not available.
[WARN] [world_model]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: . canTransform returned after 0.101041 timeout was 0.1.
[WARN] [FollowPathNode]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: . canTransform returned after 0.101109 timeout was 0.1.
[WARN] [world_model]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: . canTransform returned after 0.101169 timeout was 0.1.
[WARN] [FollowPathNode]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: . canTransform returned after 0.101188 timeout was 0.1.
```
```
$ ros2 run hz /scan

No data output.
```
```
$ ros2 run echo  /scan

No data output
```

## Feature request

#### Feature description

#### Implementation considerations

mkhansen-intel, 2018-11-30:
I haven't seen a problem where no scan data is being sent from Gazebo. What is the output in the Gazebo terminal (terminal 1 on your list above)? I suspect either a plugin is missing or not being loaded correctly.

congliu0913, 2018-12-03:
@mkhansen-intel  Below are output in the Gazebo terminal.

intel@intel-AmITX-SL-G:~/turtlebot3_ws/src/turtlebot3_simulations/turtlebot3_gazebo/worlds$ gazebo --verbose turtlebot3_ros2_demo.world -s libgazebo_ros_init.so
[Err] [ClientConfig.cc:222] Error copying default configuration file from [/usr/share/ignition/fuel_tools//config.yaml] to [/home/intel/.ignition/fuel/config.yaml]
[Err] [ClientConfig.cc:222] Error copying default configuration file from [/usr/share/ignition/fuel_tools//config.yaml] to [/home/intel/.ignition/fuel/config.yaml]
[Err] [ClientConfig.cc:222] Error copying default configuration file from [/usr/share/ignition/fuel_tools//config.yaml] to [/home/intel/.ignition/fuel/config.yaml]
Gazebo multi-robot simulator, version 9.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Gazebo multi-robot simulator, version 9.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Err] [Plugin.hh:180] Failed to load plugin libgazebo_ros_init.so: libgazebo_ros_init.so: cannot open shared object file: No such file or directory
[Msg] Waiting for master.
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.239.89.152
[Msg] Publicized address: 10.239.89.152
[Err] [Plugin.hh:180] Failed to load plugin libgazebo_ros_imu_sensor.so: libgazebo_ros_imu_sensor.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:180] Failed to load plugin libgazebo_ros_ray_sensor.so: libgazebo_ros_ray_sensor.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:180] Failed to load plugin libgazebo_ros_diff_drive.so: libgazebo_ros_diff_drive.so: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:180] Failed to load plugin libgazebo_ros_joint_state_publisher.so: libgazebo_ros_joint_state_publisher.so: cannot open shared object file: No such file or directory
[Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call

mkhansen-intel, 2018-12-03:
Looks like you don't have the ROS plugins for Gazebo. You need to clone, build and install gazebo_ros_packages on the ROS2 branch:
https://github.com/ros-simulation/gazebo_ros_pkgs/tree/ros2
```
cd navstack_dependencies_ws/src
git clone -b ros2 https://github.com/ros-simulation/gazebo_ros_pkgs.git
cd navstack_dependencies_ws
colcon build --symlink-install
source install/setup.bash
---
I think that should clone build and install everything you need. Let me know if that resolves your issue.

yechun1, 2018-12-04:
@mkhansen-intel would you please provide workable demo.rviz for gazebo? and check below topics if there is missing something.

Seems I have succeed to install gazebo, has data from /scan and /odom, but device not move after set nav goal. just like acml not work. 

here is my topic: 
```
/amcl_pose
/clicked_point
/clock
/cmd_vel
/imu
/initialpose
/joint_states
/move_base_simple/goal
/occ_grid
/occ_grid_updates
/odom
/parameter_events
/particlecloud
/robot_description
/scan
/tf
/tf_static
```

gazebo could receive rviz set nav goal.
```
$ ros2 launch nav2_bringup gazebo_rviz2_launch.py world:=turtlebot3_ros2_demo.world 
[INFO] [launch]: process[gazebo-1]: started with pid [10184]
[INFO] [launch]: process[rviz2-2]: started with pid [10185]
Gazebo multi-robot simulator, version 9.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 3 (GLSL 1.3)
[WARN] [rviz2]: Could not find folder /data/proj/ros2_ws/out_master/install/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/scripts
Gazebo multi-robot simulator, version 9.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[WARN] [rviz2]: Could not find folder /data/proj/ros2_ws/out_master/install/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/meshes
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.239.36.10
[INFO] [turtlebot3_diff_drive]: Subscribed to [/cmd_vel]
[INFO] [turtlebot3_diff_drive]: Advertise odometry on [/odom]
[INFO] [turtlebot3_diff_drive]: Publishing odom transforms between [odom] and [base_footprint]
[INFO] [turtlebot3_joint_state]: Going to publish joint [wheel_left_joint]
[INFO] [turtlebot3_joint_state]: Going to publish joint [wheel_right_joint]
[INFO] [rviz2]: Creating 1 swatches_
[INFO] [rviz2]: Setting goal: Frame:map, Position(-0.742419, -2.17702, 0), Orientation(0, 0, 0, 1) = Angle: 0
[INFO] [rviz2]: Setting goal: Frame:map, Position(0.208433, -1.96276, 0), Orientation(0, 0, 0, 1) = Angle: 0
[INFO] [rviz2]: Setting goal: Frame:map, Position(-0.758896, -2.44903, 0), Orientation(0, 0, 0.413098, 0.910687) = Angle: 0.851706
```

mkhansen-intel, 2018-12-04:
It looks like Gazebo and Rviz are running correctly now. Do you have the nav2_bringup_1st_launch.py or nav2_bringup_launch.py running in another window? If not you need to launch that according to our bringup README here:
https://github.com/ros-planning/navigation2/blob/master/nav2_bringup/README.md

If they are launched, what output do you see in that window when you click on '2D Pose Estimate' button to localize the robot? You should place the arrow at the same location in the Rviz map that the robot is located in the Gazebo world.

mkhansen-intel, 2018-12-05:
@yechun1 - I'm glad you have it working. Please do submit a PR to improve the bringup. Let me know if you need more help. I'm closing this issue now.

closed 2018-12-05 by mkhansen-intel