ros-navigation/navigation2 #401: Add support for Keep-out zones on map

mkhansen-intel, 2018-12-03:
## Feature request
Keep out zones
#### Feature description

User should be able to specify keep-out zones on a map via a GUI and/or via a file (XML? YAML?)
Keep out zones can be limited to rectangles to begin with

#### Implementation considerations

Should be possible to do this with a costmap layer, but planner support may be required also

SteveMacenski, 2019-12-06:
@mkhansen-intel this is covered by #1263

mkhansen-intel, 2019-12-06:
I don't think so, that says traffic lanes in the description.

SteveMacenski, 2019-12-06:
I think the discussion in the ticket describes a path forward that makes them all the exact same problem description, with a single operational difference.

This to me then signals this is a great candidate for a generic costmap layer with plugins for how to handle cost. Which is the same as traffic lanes, keep out, slow down, etc

crdelsey, 2019-12-09:
> traffic lanes, keep out, slow down.

There are differences in each. I don't believe they can all be handled exactly the same way.

- Keep Out Zones - These can be simply represented by an extremely high cost in the costmap.
- Slow Zones - These probably get represented as a slightly higher cost in the costmap to encourage avoiding them if possible, but there is additional work to change the max speeds in DWB - using either a new critic or having something tweak DWB max velocity parameters.
- Traffic Lanes - probably can be represented in the costmap, but there seems to be a corresponding change in the planner needed. The cost of a traffic lane depends on the direction of the plan at that point.

SteveMacenski, 2019-12-09:
Do you have a few minutes to call me?

SteveMacenski, 2020-07-01:
#1263 covers and underway

closed 2020-07-01 by SteveMacenski