ros-navigation/navigation2 #404: Add Slow / safety zones to map

mkhansen-intel, 2018-12-03:
## Feature request
Slow / safety zones
#### Feature description

User should be able to specify slow/safety zones on a map via a GUI and/or via a file (XML? YAML?) Zones can be limited to rectangles to begin with

#### Implementation considerations

Should be possible to do this with a costmap layer, but planner support may be required also

SteveMacenski, 2019-05-21:
I believe that this ticket https://github.com/ros-planning/navigation2/issues/517 is a subset of the intention of this one: adding safety sensing ability to a reactionary stop for an imminent collision. 

I'm working on this now. To start it will support just laser scanners, but after I have some other folks look at it, I will extend to cover sonars and bump sensors as well to cover the bases of all the "safety" grade robotics sensors.

SteveMacenski, 2019-12-06:
This is covered by #1263

mkhansen-intel, 2019-12-06:
I don't think so, that says traffic lanes in the description.

SteveMacenski, 2019-12-06:
I think the discussion in the ticket describes a path forward that makes them all the exact same problem description, with a single operational difference.

This to me then signals this is a great candidate for a generic costmap layer with plugins for how to handle cost. Which is the same as traffic lanes, keep out, slow down, etc

mkhansen-intel, 2019-12-06:
I don't disagree that the implementations may be similar and even overlapping, however the 'feature' from a users and robots perspective is different. This is why they have separate issues filed.

SteveMacenski, 2019-12-06:
I don't think they should be separate issues because of how closely they relate that in the process of implementing 1, you should implement them all since they share 90% of the code. This also makes [whomever] is doing that at the time think about design so that these are extendable and not just 1-offs. 

I think adding a costmap-filter-like plugin interface that can be used to make these features would be and invaluable asset. Then as there are more tasks for spatial-setting-scheduling all the tools are there already. I doubt you and I could come up with an exhaustive list of all the things someone might want to do with this kind of interface so its more important to me to have it built in a way that is extendable like the costmap-plugins for new usecases. This is the main reason I put them in the same ticket so that someone wouldn't just make a one-off executable or layer

SteveMacenski, 2020-07-01:
#1263 covers and underway

closed 2020-07-01 by SteveMacenski