ros-navigation/navigation2 #406: Add clear costmaps recovery

SteveMacenski, 2018-12-03:


SteveMacenski, 2019-02-01:
clear costmap is a recovery behavior, not itself a layer, so it should not go into costmap_2d, its a recovery plugin. Move base also exposes a clear costmaps service that people also typically used to clear costmaps so you dont have to get to a recovery state before calling it. 

Having one of the recovery actions be clearing costmaps makes sense to me. Costmap2D is just the perception items to present and be used by a higher level orchestrating piece of software, that thing should contain any additional logic you'd like to add in recovery, not inside the costmap itself. Each layer and layeredcostmap has the reset method to reset "clear" them

orduno, 2019-02-01:
To me it makes sense that `Costmap2D` should have a `clear()` method, where the layers and regions to reset are passed. I don't think of this as a behavior, but a feature of `Costmap2D`. Just as I would expect other representations to have an interface for clearing its underlying structure.

It then becomes a behavoir when the higher level orchestrating software, for some reason, clears the world representation.

SteveMacenski, 2019-02-01:
This functionality exists in Costmap2D https://github.com/ros-planning/navigation/blob/melodic-devel/costmap_2d/include/costmap_2d/costmap_2d.h#L341 and https://github.com/ros-planning/navigation/blob/melodic-devel/costmap_2d/include/costmap_2d/costmap_2d_ros.h#L112

Costmap_2d uses the "reset" word rather than "clear"

orduno, 2019-02-01:
@SteveMacenski Clearing/resetting with those methods clear all the layers entirely.

I discussed with @crdelsey his proposal of moving the behavior to Costmap2DROS and adding service for it. This is a short term solution until we move the local costmap to world model.

I think we need to do this in the following steps:
- [ ] Move the behavior to Costmap2DROS and add service #551.
- [ ] Consolidate costmaps to World model #550.
- [ ] Revert #551 from Costmap2DROS. Some functionality will go to Costmap2D, i.e. method for clearing a particular region on a specific layer. Other to World Model.
- [ ] Define the situation to detect for triggering the clear costmap behavior. Implement the logic on the BTNavigator, create the corresponding BT nodes #552.

orduno, 2019-06-10:
The three types for clearing a costmap have been added.
In addition, `navigate_w_replanning_and_recovery.xml` behavior tree runs these services during recovery.

@SteveMacenski since you created the issue, do you think we have addressed all the concerns?

closed 2019-06-21 by orduno

SteveMacenski, 2019-07-11:
@orduno I was looking for more of an official clear costmap recovery that may call the clear costmaps service over [this](https://github.com/ros-planning/navigation2/blob/master/nav2_bt_navigator/behavior_trees/navigate_w_replanning_and_recovery.xml#L17) in the BT that seems less trivial to find. 

I imagine that the motion_primitives package will turn into the recoveries package so having the non-motion recovery is totally valid.