ros-navigation/navigation2 #417: Occasional BT Navigator segfault

mjeronimo, 2018-12-05:
Seems to happen in the spin loop at shutdown, but more investigation needed to characterize the problem.

orduno, 2018-12-07:
I had multiple segfaults yesterday, all pointing to the DDS layer. I wonder if it's an issue of the particular DDS implementation we're using (eProsima Fast RTPS).

At some point we should test [other implementations](https://index.ros.org/doc/ros2/DDS-and-ROS-middleware-implementations/). 

On a related topic, when debugging we can also use the [DDS tools and ecosystem](https://www.rti.com/blog/ros2-dds-when-ecosystems-merge).

orduno, 2018-12-09:
Here's a different segfault.
[image]

crdelsey, 2019-01-28:
Enable BT by default

crdelsey, 2019-02-01:
This segfault doesn't seem to occur anymore. We'll close this after more testing.

mkhansen-intel, 2019-02-15:
Can we close this now?

closed 2019-02-15 by crdelsey