ros-navigation/navigation2 #421: Convert new development efforts to newly created crystal-devel

SteveMacenski, 2018-12-05:
This is the general ROS naming to work on software inside the ROS ecosystem so that when new releases come out (that usually break sections of API) you still have all the code to support the last versions.

We should change default branch from `master` to `crystal-devel` now that we're working on a release-able distro.

mkhansen-intel, 2018-12-07:
I would prefer to do this only when necessary, meaning only if a breaking API change is required that can't be made backwards compatible. Maybe we can talk about this in the next Nav2 WG? In general the more branches you have to maintain, the more work it is just to keep things sync'd.

SteveMacenski, 2018-12-07:
@mkhansen-intel You're still only maintaining 1 branch: the current devel branch. Generally speaking once a new distro comes out you stop developing on the old branches and only cherry pick commits to go backwards when there's a serious flaw that's resolved, but no new features.

Example:
D release comes out, crystal-devel stays in place, but new PRs are directed to D-devel branch. Only if something super significant and needed to go backwards is fixed would you update crystal-devel, otherwise you leave the crystal-devel branch as no longer under active development. This is how navigation1 is currently maintained and how most ros packages/stacks are maintained as far as I know.

Leaving it as `master` doesnt tell a user which distro this will work on, and if there's an API change making it not work in - lets say bouncy (which is true), then they're in some trouble and need to find the last commit where bouncy still works, which is border line impossible in many situations

jbvo, 2018-12-07:
There was a discussion about this practice on Discourse a few months ago: https://discourse.ros.org/t/on-git-branching-strategies/4785.

As a user of ROS, I personally find the use of the `*-devel` branches rather confusing, especially the part where `master` is not the default branch.
In nearly all other open source projects that I have used or contributed to, it was common to create a maintenance branch for older releases for the purpose of developing and releasing (critical) bugfixes. Any other development usually would target `master` (Github or Gitlab flow) or in case of git flow it targets `develop`.

SteveMacenski, 2018-12-07:
Usually in these projects, you delete the master branch, there isn't one. If there is one, you should probably be ignoring it.

The problem if we release new versions so frequently you can't keep tagging one branch to do releases on easily, they need to be separate branches so I can release them in bloom as zips for the release repositories to use. It needs to be immediately clear to users which distro master represents. You assume a perfect world where everyone looking at the project understands what distrobution they're possibly looking at and that they have the same install - it needs to be explicit or else we're going to be fielding 10 questions a week on issues/ros answers saying it doesnt compile for them. 

This also assumes this project keeps up to date with all the releases, which I would not assume out of the gate - often folks skip a cycle if its a minor release without LTS support and there's not much active development going on. 

Anyhow, I'd really like to hear from the intel folks what you think should happen. We need separate branches to support separate distros to be able to release. It's more of a question whether we should work off of a master or a devel branch. I think devel branch (or even if we just called it `crystal` without devel) makes sense and makes it incredibly explicit about what's happening for a on-looker. Also as a ros community we've been doing this for awhile and I don't see that this project espectially should break rank from that due to its importance in the ros ecosystem.

mkhansen-intel, 2018-12-07:
I read the Discourse thread and I'm personally a [Trunk Based Development](https://trunkbaseddevelopment.com/) believer. The diagram below from the Discourse thread is what I had in mind. 
```
C3' <- C6' <- [kinetic]    [lunar]     [melodic]
 /                        /           /
C1 <- C2 <- C3 <- C4 <- C5 <- C6 <- C7 <- [master]
```
In this model we:
* Always do pull requests to `master` for features
* If there's a breaking API change, we tag the last commit before it with the release it supports
Example - if we break API compatibility with Crystal at some future date, we would tag the last commit in master before that as `crystal` so that anyone using Crystal could check out the tag to build for Crystal. (This is like the `[lunar]` and `[melodic]` tags in the diagram above.)
* If after we've tagged it and gone forward with the breaking API on master, we need to fix a bug in a prior release, THEN we create a branch starting with the Crystal tag, add the bug fix and move the tag to match the branch HEAD. (This is like the `[kinetic]` branch above)
* This way we avoid branching until it is absolutely necessary and we only do it for bug fixes

That's my opinion, but I'm willing to discuss. I think the model I described, although different from the ROS model, is more consistent with other open source projects.

SteveMacenski, 2018-12-07:
@mkhansen-intel lets chat next week at the WG meeting - I think this would create alot of additional stress on maintainers to cherry pick backwards vs just have the branches going with them. I'm a believer that while the distro is under ROS LTS we need to support them with updates regularly. Even though Kinetic in ROS1 isn't the newest kid on the block, all major fixes are cherry picked from Melodic back down to Kinetic because its still under LTS support and these bug fixes still effect many number of users. It will be pretty regular for us to cherry pick commits back into old releases in (what I'm going to guess is) 30% of PRs. 

If we want to use `master` as the development branch, I can get behind that I suppose but then there needs to be something clear somewhere on the landing page for the repo that master = {crystal, D-release, etc}.

mkhansen-intel, 2018-12-08:
@SteveMacenski - sure we can talk about it. One thing that comes to mind that should be considered as part of this discussion is that right now Crystal is not termed an LTS by OSRF. I don't think they've even reached a level of maturity yet to say any ROS2 release is an LTS. Maybe the D-Turtle release will be an LTS, but that hasn't been discussed publically AFAIK. So I don't expect when we move to D-Turtle for Nav2 we will backport *any* bug fixes to Crystal, partly due to OSRF lack of LTS and partly due to the fact that our first release is really 'experimental', not LTS. Anyway, we need to consider this too in the discussion.

SteveMacenski, 2018-12-08:
That's a good point. It's just harder to change habit 12-24 months from now once other people rely on it and get used to a certain structure. We can readdress then but it'll be harder to get the boat rocking

crdelsey, 2019-02-12:
@SteveMacenski I just updated the `crystal-devel` branch with all the latest changes we want off `master`.  I'd like to get the OSRF build farm building this branch instead of master. I'm thinking we can call this a 0.1.6 version as a way trigger the update.

What's the process for doing this? Is it:
1. Bump all the version numbers on the crystal-devel branch and tag it.
2. Update the version number in https://github.com/SteveMacenski/navigation2-release/blob/master/tracks.yaml
3. Run the bloom command you have listed and push the resulting PR ros/rosdistro?

Can you help with this?

SteveMacenski, 2019-02-13:
So you dont want the build farm PR builder on master anymore? that seems like it could be dangerous as we're doing development work but not aware of hard API changes.

Assuming you guys have thought about this and are sure, I'd be happy to, just follow up  that we're all on the same page or if we want to setup a conference call

crdelsey, 2019-02-13:
@SteveMacenski Are you able to attend this Thursday's WG meeting? We could discuss it then. Some folks are away until then anyhow. Otherwise we can arrange an alternate time.

The situation now, though, is that our master branch is following the OSRF master, which has just made an API change for Dashing. Our master is no longer compatible with Crystal anymore, so the build farm PR builder is always going to report a build failure. We need to do something to get the build passing, but maybe what I suggested is not the right thing.  Maybe we can still make master build for both releases.

SteveMacenski, 2019-02-14:
I plan to be there, yes. We can discuss then and I'll update with the decision made

SteveMacenski, 2019-02-14:
Moving PRs and docs over to crystal: https://github.com/ros/rosdistro/pull/20317

I'll sync against crystal when we're doing porting over stuff we want

SteveMacenski, 2019-02-14:
I think this discussion also pretty much summed up the scope of the ticket, so I don't think this is needed anymore.

closed 2019-02-14 by SteveMacenski