ros-navigation/navigation2 #425: Robot stabilization on recovery behavior

orduno, 2018-12-05:
When the robot bumps into objects it might tilt or oscillate. For now, the `Stop` motion primitive waits for a fixed amount of time to allow for it to settle down, however, we might want to actively monitor this state and vary the wait time. Perhaps also do active stabilization.

mkhansen-intel, 2018-12-07:
Most robots have an IMU so you can use the accelerometer and gyro to make sure the robot motion has stabilized. by subscribing to the IMU topic and looking at the acceleration values in linear and angular terms. If an IMU isn't available, then default to a time-delay as a backup case.

crdelsey, 2019-07-01:
This could be a pause type BT node or a node that just waits for IMU stabilization

SteveMacenski, 2019-09-04:
Is this still a ticket to review? The stop recovery is gone

mkhansen-intel, 2019-09-11:
I think we should close it

closed 2019-09-11 by mkhansen-intel