ros-navigation/navigation2 #436: Localization system test fails 24/200 times

mkhansen-intel, 2018-12-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - master branch: https://github.com/ros-planning/navigation2/commit/bd12f5b3b8f5db8fa2ce365a53d0aefe73a3426f
- DDS implementation:
- default 
- Client library (if applicable):
-  rclcpp

#### Steps to reproduce issue

Run ctest in a loop using a bash script:
```
fail=0
total=$1
# Change the file path manually to where you want to log fails
file=/home/mkhansen/ros2_dev/navigation2_ws/temp/fails.txt
echo "Total Loops = " $total
for ((i=1;i<=total;i++))
  do
    ctest
    result=$?
    echo "RESULT =" $result
    if [ "$result" != "0" ]
    then
      ((fail+=1))
      echo "TEST $i FAILED" >> $file
    fi
    echo $i "LOOPS COMPLETED"
    echo $fail "TOTAL FAILURES"
  done
echo $fail " LOOP FAILURES / " $total
echo $fail " LOOP FAILURES / " $total >> $file
```

Run it from the directory you normally run ctest from. It takes one parameter right now, the number of loops to run. Example:
```
$ source ctest_loops.bash 50
```
I ran the localization test 200 times with 24 fails.

#### Expected behavior
Ideally, the test should pass every time
#### Actual behavior
It fails at about a 12% rate
#### Additional information

I then made this change to retry if the result was bad. I know it doesn't restart everything but it resends the initial pose and re-checks the result.
```
mkhansen@mkhansen-desk:~/ros2_dev/navigation2_ws/src/navigation2$ git diff nav2_system_tests/src/localization/test_localization_node.cpp
diff --git a/nav2_system_tests/src/localization/test_localization_node.cpp b/nav2_system_tests/src/localization/test_localization_node.cpp
index b76d8c00..9ad42c72 100644
--- a/nav2_system_tests/src/localization/test_localization_node.cpp
+++ b/nav2_system_tests/src/localization/test_localization_node.cpp
@@ -111,5 +111,11 @@ void TestAmclPose::initTestPose()
 
 TEST_F(TestAmclPose, SimpleAmclTest)
 {
-  EXPECT_EQ(true, defaultAmclTest());
+  bool result = defaultAmclTest();
+  if (!result) {
+    // retry test one more time
+    result = defaultAmclTest();
+  }
+
+  EXPECT_EQ(true, result);
 }
```
The test still fails at the same rate (24/200). Changing the tolerance values from 0.25m to 0.5m may enable it to pass more often.

mkhansen-intel, 2019-02-15:
@mhpanah - was this fixed?

crdelsey, 2019-02-15:
This isn't fixed. `test_localization` and `test_system_node` are still flaky. #529 just got the tests back from 100% failure to the typical 30% failure.

mkhansen-intel, 2019-02-15:
@mhpanah - at one point you did some work on this and at least had data for test_localization. Can you comment as to the status of this?

mhpanah, 2019-02-19:
> @mhpanah - at one point you did some work on this and at least had data for test_localization. Can you comment as to the status of this?

Will modify the test to use rosbag.

mkhansen-intel, 2019-09-18:
This is out of date, closing.

Current results and bugs are being tracked in other issues like #1129

closed 2019-09-18 by mkhansen-intel