ros-navigation/navigation2 #439: Future of dependencies.repos file

SteveMacenski, 2018-12-07:
We have some options here. We can continue to have this file for things like behavior trees, or other repositories that will be under development that we want to build locally with on bleeding edge. Things like angles/message filters should be removed and build with stable once the crystal sync comes around.

Once the sync comes because all of our dependencies are met, we can remove the repos file altogether *or* maintain it so we can have a few repositories building locally on their bleeding edge masters.

Thoughts?

mkhansen-intel, 2018-12-07:
tl;dr: While I agree we want to remove it, I don't think we can do that yet.

Ideally, I agree with you we would like to remove it. However, I'm not sure that we'll have our dependencies met for awhile if we're developing for the 'next' ROS2 release. We may have forks of upstream projects with bug-fixes and work-arounds (like we had with launch for awhile with the timer fix). So I think it's too early to eliminate it completely.

One area we may have dependencies is for the system tests. Right now my branch and the localization test are dependent on gazebo_ros_pkgs changes that aren't released yet, and similarly turtlebot3 packages, those may not be released to debian for quite awhile. Ideally we'd like to remove dependencies on anything that's not released in debian, but it may take time.

SteveMacenski, 2018-12-07:
I think that all makes sense, if we're relying on upstream dependencies that we might need to patch, we should leave them. Just wanted to get the thought out there. Closing ticket.

closed 2018-12-07 by SteveMacenski