ros-navigation/navigation2 #461: Nav2 Tasks tests has a memory issue needing resolved for crystal release

SteveMacenski, 2018-12-09:
```
18:03:10 1: Test command: /usr/bin/python3 "-u" "/opt/ros/crystal/share/ament_cmake_test/cmake/run_test.py" "/tmp/ws/test_results/nav2_tasks/test_tasks.gtest.xml" "--package-name" "nav2_tasks" "--output-file" "/tmp/ws/build_isolated/nav2_tasks/ament_cmake_gtest/test_tasks.txt" "--command" "/tmp/ws/build_isolated/nav2_tasks/test_tasks" "--gtest_output=xml:/tmp/ws/test_results/nav2_tasks/test_tasks.gtest.xml"
18:03:10 1: Test timeout computed to be: 60
18:03:10 1: -- run_test.py: invoking following command in '/tmp/ws/src/navigation2/nav2_tasks':
18:03:10 1:  - /tmp/ws/build_isolated/nav2_tasks/test_tasks --gtest_output=xml:/tmp/ws/test_results/nav2_tasks/test_tasks.gtest.xml
18:03:10 1: Running main() from gtest_main.cc
18:03:10 1: [==========] Running 1 test from 1 test case.
18:03:10 1: [----------] Global test environment set-up.
18:03:10 1: [----------] 1 test from TaskClientServerTest
18:03:10 1: [ RUN      ] TaskClientServerTest.allTests
18:03:12 1: [       OK ] TaskClientServerTest.allTests (2112 ms)
18:03:12 1: [----------] 1 test from TaskClientServerTest (2112 ms total)
18:03:12 1: 
18:03:12 1: [----------] Global test environment tear-down
18:03:12 1: [==========] 1 test from 1 test case ran. (2112 ms total)
18:03:12 1: [  PASSED  ] 1 test.
18:03:50 1: double free or corruption (fasttop)
```

bpwilcox, 2018-12-10:
I've been getting the same error just recently when running the nav2_costmap_2d tests in #387:
`double free or corruption (fasttop)`
I'm not yet sure how to resolve it, which is why I haven't merged. I might need to break into error with debugger tools.

SteveMacenski, 2018-12-10:
We can get away with it for right now* as I fixed all the other tests in my PR (https://github.com/ros-planning/navigation2/pull/458 which I need approved ASAP to get rolling on the crystal release) but that should be resolved long term

mkhansen-intel, 2019-02-15:
@SteveMacenski - is this resolved or still an open issue that needs fixing?

crdelsey, 2019-02-15:
So far as I know this is still a problem. I believe we commented out the test to get things to pass. I was planning to look into this and the other unit test issues this month.

crdelsey, 2019-07-01:
Tasks and this test are gone now. Closing.

closed 2019-07-01 by crdelsey