ros-navigation/navigation2 #466: map_saver saves hard-coded values for free and occupied thresholds

mkhansen-intel, 2018-12-10:
## Bug report
The code in the ROS navigation map_saver currently saves hard-coded values for free threshold and occupied threshold. This essentially ignores what was passed in on the command line.

mkhansen-intel, 2018-12-13:
As part of this bug fix, we should determine whether the threshold values should be parameters or arguments as they are today.

crdelsey, 2019-01-22:
This has been around for a long time, so making it low priority for now.

rotu, 2019-07-23:
Additionally, I would like to get rid of thresholds altogether and store the actual OccupancyGrid values. That way I can do my own tracing and/or cleanup on the images from SLAM.

mkhansen-intel, 2019-07-23:
@rotu - would support for SCALE and RAW modes meet your needs (as defined in map_loader.hpp)

```
/** Map mode
 *  Default: TRINARY -
 *      value >= occ_th - Occupied (100)
 *      value <= free_th - Free (0)
 *      otherwise - Unknown
 *  SCALE -
 *      alpha < 1.0 - Unknown
 *      value >= occ_th - Occupied (100)
 *      value <= free_th - Free (0)
 *      otherwise - f( (free_th, occ_th) ) = (0, 100)
 *          (linearly map in between values to (0,100)
 *  RAW -
 *      value = value
 */
```

This means you could run `map_saver --mode RAW` to do what you're asking

rotu, 2019-07-24:
@mkhansen-intel I think so. I'm working on the PR right now, actually.

rotu, 2019-07-24:
@mkhansen-intel Here's where I am:
https://github.com/ros-planning/navigation2/blob/a2074b639a10e60c6a48116d7565e01782e01b40/nav2_map_server/src/map_saver.cpp

It doesn't totally work yet. One of the problems is I can't figure out how to get SDL (image library used in map loader) to write any image format other than bmp (including pgm). 

Here's what I'm thinking as far as options:
1. Keep using pgm. Admit that we can't use SCALE mode because pgm has no alpha channel.
2. Accept BMP as our new writing format and use SDL image library for writing.
3. use a different library: CImg: lightweight image library but weird license, ImageMagick: fully functional weird license, OpenCV2: heavy library but already used in ROS2 stack
5. port this node over to Python and use Pillow

Advice?

SteveMacenski, 2019-07-24:
I would say change libraries over change file types, that way people like a long history of maps don't have a heart attack and we don't have to provide any conversion tools. I don't know that anyone has any love for a particular image IO library. 

Also wanted to say that I love the save/stop by occupancy value, I think there's only 1 or 2 SLAM options that could make total use of it but that's a great direction to start pushing.  

> ... can't use SCALE

I can tell you in map server ROS1, that scale definitely works. I have used it for... things in the past. With that said, if changing libraries makes your life easier, power to you. 

Python would not be great - I think this is a particular node that is a good candidate for python but by in large python in production is something to be avoided.

rotu, 2019-07-25:
Thanks, @SteveMacenski, for the feedback.

> I would say change libraries over change file types ... I don't know that anyone has any love for a particular image IO library.

Sounds good to me. I'm going with ImageMagick.

> Also wanted to say that I love the save/stop by occupancy value

I don't understand what you mean. Rephrase?

> I can tell you in map server ROS1, that scale definitely works.

Scale works for `map_server` but not `map_saver`. It relies on having a separate alpha channel, and PGM has just one value for pixel (correct me if I've missed something). You *can* use PNG with an alpha channel, but the problem is that while the SDL library can read such files, it cannot write them.

SteveMacenski, 2019-07-25:
Sorry, rephrase is:

> thanks for doing this, I agree

Ah, you are right. I modify maps with my own toolset not through map saver.

rotu, 2019-07-25:
Development is coming along at a healthy clip:
https://github.com/RoverRobotics-forks/navigation2/commit/3d7382358ddd58c7672ac41d42833fc1742f2bac

SteveMacenski, 2019-09-04:
@rotu's PR resolves

closed 2019-09-04 by SteveMacenski