ros-navigation/navigation2 #468: Add validation for all dynamic parameters

bpwilcox, 2018-12-10:
Currently in the stack, most dynamic parameters are added to the `DynamicParamsClient` without any required validation when set dynamically. This could lead to erroneous or unsafe behavior if set to an extreme value or a different data type. Thus, dynamic parameters should be added to a `DynamicParamsValidator`  and given appropriate validation requirements.

mkhansen-intel, 2019-02-19:
This should be done after lifecycle nodes are implemented. Care should be taken as to which lifecycle state a given parameter can be changed in. Most parameters should probably be set as part of the configuration step, then all changes to them should be ignored while in the active state. Only a small subset should be allowed to be changed when in the active state.

bpwilcox, 2019-10-01:
The `nav2_dynamic_params` package has been removed, so this would apply to the re-integration of dynamic parameters. I'll close this issue and this can be apart of a new design issue

closed 2019-10-01 by bpwilcox