ros-navigation/navigation2 #477: Update exported dependencies

orduno, 2018-12-11:
Meta-ticket for auditing the nav2 packages looking for [unnecessary exported dependencies](https://github.com/ros-planning/navigation2/search?q=ament_export_dependencies&unscoped_q=ament_export_dependencies).

- [ ] nav2_robot - @mhpanah 
- [ ] nav2_dwb_controller - @crdelsey 
- [ ] nav2_bringup - @mkhansen-intel 
- [ ] nav2_voxel_grid  - @bpwilcox 
- [ ] nav2_msgs - @mjeronimo 
- [ ] nav2_tasks - @orduno 
- [ ] nav2_amcl - @mhpanah 

Checkout #474 for an explanation on when to use `ament_export_dependencies`.

mkhansen-intel, 2018-12-11:
@orduno - shouldn't rosidl_default_generators be removed as a dependency from nav2_tasks?

crdelsey, 2018-12-11:
There are others that should be added back in to nav2_tasks as well. For example, std_msgs, nav2_msgs. Maybe others.

mkhansen-intel, 2018-12-11:
I added #478 for nav2_bringup

SteveMacenski, 2018-12-11:
To clarify, we **do not have to get rid of redundant or unused dependencies** we just need to not **export** things that break isolated builds

mkhansen-intel, 2019-02-15:
Is this still an issue or has it been resolved?

crdelsey, 2019-02-15:
I think we can close this. #496 would detect the problem that this issue raises.

closed 2019-05-17 by orduno