ros-navigation/navigation2 #482: Update BTs with recovery

orduno, 2018-12-12:
[See details here](https://github.com/BehaviorTree/BehaviorTree.CPP/issues/36)

orduno, 2018-12-19:
Bellow is our current behavior tree with simple navigation and recovery and @facontidavide 's proposal

current | proposal
--- | ---
[image] | [image]

So far I see two issues:
- We need two `IsStuck` nodes. While the first `IsStuck` node get's ticked periodically, the second doesn't. If these two nodes are independent, it's possible that one may detect that the robot is stuck while the other not. This could be potentially fixed by shared the state throught the BT::Blackboard.
- As shown, the navigation doesn't resume once the `stuck_recovery` BT subtree is executed.

facontidavide, 2018-12-19:
Good points. 

I am not particularly worry about the first one, since there are multiple way to avoid issues there.

Focusing on the second one...

I have few ideas about how this can be solved. For example adding a __RetryUntilSuccesfull__ decorator to resume the entire tree after a recovery.

But this trigger a more profound question. what do you really want to happen?

- Do you want `BackUp` and/or `Spin` to be interrupted if  `IsStuck` becomes suddently true or do you want them to complete their routine?

- How many time do you want to resume `navigation_subtree` after a `stuck_recovery`? A limited amount of times? 

The more details you give me about the desired behavior, more I can help ;)

Cheers

orduno, 2019-06-10:
Closing the issue as we have a new behavior tree for robot recovery `navigate_w_replanning_and_recovery` that fixes this.

FYI, the logic on the new bt tree is different, we no longer check for preconditions and only do recovery if planning or control fails.

closed 2019-06-10 by orduno