ros-navigation/navigation2 #496: CI should build the navstack in the same workspace as ROS2 and dependencies

crdelsey, 2018-12-17:
## Feature request

#### Feature description

The current directory structure is:
* Top Dir
  * ROS2 Workspace
  * Nav2 Dependencies Workspace
  * Nav2 Workspace
Each workspace is built independently.

Instead, if we restructure it like this:
* ROS2 Workspace
  * src
    * Existing ROS2 directory structure
    * nav2 dependencies
    * navigation 2

And then just run `colcon build` on the ROS2 workspace, this will catch more dependency errors. See #473.

crdelsey, 2019-03-18:
Related to #606

crdelsey, 2019-07-01:
Dockerfile to run this test is available. It hasn't been integrated into CI.

ruffsl, 2019-07-25:
> It hasn't been integrated into CI.

Is there any CI option that wouldn't time out from that length of compilation time?

SteveMacenski, 2019-09-04:
Given the current work you've done (thanks!) @ruffsl is this still something we should do?

ruffsl, 2019-09-04:
In https://github.com/osrf/docker_images/issues/294 I suggest the idea of having a docker image for source build, where everything would be in one single workspace. I'm not sure we could have one CI job build everything in one go, given the time out length of most CI platforms. I was thinking we could front load it across two separate image builds. But no one has replied to the idea. However, is this something that ci.ros2.org would be able to do better?

SteveMacenski, 2019-09-04:
If we do it across multiple build images, do you know if dockerhub's timeouts wouldn't be enough? I believe they're 2 hours. 

Do you think this is still the direction we should go for CI or is what we have is sufficient?

ruffsl, 2019-09-04:
If we kept caching via docker images, then across multiple build images would be the way. I'll have to test if something like `osrf/ros2:source` times out or not on docker hub. If not, then something like `rosplanning/navigation2:master.source` that would build `FROM` `osrf/ros2:source` should be fine, as that's not nearly as long of a build.

SteveMacenski, 2019-09-04:
> However, is this something that ci.ros2.org would be able to do better?

I think that's a definitely "yes". What's the difference between the nightly, source, and devel images on ROS2 master? 

I guess maybe I'm missing something, the `rosplanning/navigation2:master.source` would be good for deployment and daily syncing, but how does that help with CI, or is that building with each PR?

ruffsl, 2019-09-04:
> What's the difference between the nightly, source, and devel images on ROS2 master?

I've added to the repo readme here: https://github.com/osrf/docker_images/pull/321 . We'd simply have our CI build `FROM` the `osrf/ros2:source` image to avoid timing out our CI.

> but how does that help with CI, or is that building with each PR?

We could trigger the ci to build the Dockerfile that generated the `rosplanning/navigation2:master.source` image. It wouldn't need to push it anywhere, just be sure that it builds. Sort of like what the travis CI is up now. I was thinking we could use `rosplanning/navigation2:master.source` to cache the incremental build, sort of like what we do now for debug and release, but perhaps a fresh docker build each git push would be more thorough.

SteveMacenski, 2019-09-04:
So something like this would replace travis, but not CircleCI? Its the full nightly build vs. incremental cache? 

I suppose maybe some of these questions come from not really understanding why we have so many build agents. Why is something like 
- takes nightly image
- new nightly image triggers dockerhub build of just the navigation2 dependencies repo file
- PRs pull/cache this image and use to build
- Merging in master triggers travis or circle to do a clean from scratch build

not possible?

ruffsl, 2019-09-04:
> So something like this would replace travis, but not CircleCI?

Eh, we could use travis to `docker build` the PR, and that respective Dockerfile would build the PR branch `FROM` `osrf/ros2:source` in the same monolithic workspace to check dependency declaration.

> Its the full nightly build vs. incremental cache?

We'd be using `osrf/ros2:source` as a cache if you will, so we don't have start from scratch for each pr.

Hmm, perhaps I could better explain the current CI pipeline over the meeting time? I've been meaning to write a blog post documenting all the directions and design decisions, but I've been short on time.

SteveMacenski, 2019-09-04:
Yeah that would be great!

crdelsey, 2019-09-04:
@ruffsl All this work might be worth it if multiple projects can use this. Now that I've got a dockerfile to run this test, it's relatively easy to run this manually before doing a release. I've kind of ignored this issue because of that.

mkhansen-intel, 2019-09-11:
@ruffsl - if you can point me to the osrf/ros2:source I could modify our dockerfile to use it and see how long it takes in Travis CI. I don't see it on dockerhub.

ruffsl, 2019-09-13:
> can point me to the osrf/ros2:source

@mkhansen-intel , Its in the ros2/source folder:
https://github.com/osrf/docker_images/blob/e8f80e9bdc35dff2cfe4a114d72fece8d23fb286/ros2/source/source/Dockerfile

I've started a draft PR here you cant look at: https://github.com/ros-planning/navigation2/pull/1119

closed 2019-10-12 by SteveMacenski