ros-navigation/navigation2 #517: Check for free space directly from laser scan

orduno, 2019-01-08:
In world model, enable a service for checking for free space directly from laser scans instead of costmap, with region specified relative to the robot.

See #516 for more background info.

SteveMacenski, 2019-01-08:
Can you explain the motivation?

SteveMacenski, 2019-01-08:
> On a low-level safety reactive layer, to avoid a collision as a reactive measure.

I think this should be a server or plugin interface where one implementation of it could be looking at a laserscan. Some robots don't have lasers or lasers appropriate for that application due to resolution, size, or placement. 

Checking the costmap lets you fuse in other sensors (multiple lidars, radar, sonar, depth, etc) which should be generally used. If you're trying to make some type of collision zone type of implementation, I'd make sure its configurable for multiple robot setups but generally still advise to use the costmap unless you're wanting to query this at 100Hz or something

crdelsey, 2019-01-09:
Also, for a low-level safety check, I'd expect you'd want it isolated in a separate process from the potentially complex world model.

Can you explain how this relates to self-localization?

crdelsey, 2019-07-03:
I'd like to close this issue. I think we no longer need this now that we can run costmaps in the odom frame.

closed 2019-07-08 by crdelsey