ros-navigation/navigation2 #520: DWB oscillation detection seems overly agressive

crdelsey, 2019-01-08:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - Master as of 8 Jan 2019
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Description

When the robot is travelling around the map, especially at the beginning or ending of a path as the robot tries to rotate to the correct position, the DWB controller will fail indicating

```
[ERROR] [DWBLocalPlanner]: No valid trajectories out of 399! 
[ERROR] [DWBLocalPlanner]: 1.00: Oscillation/Trajectory is oscillating.
[INFO] [DwbController]: No valid trajectories out of 399! 
[ERROR] [simple_navigator]: Control task failed.
```

It seems unreasonable that every possible trajectory results in oscillation.

crdelsey, 2019-01-22:
This problem is aggravated by debug builds, where the DWB implementation doesn't run fast enough to result in stable feedback, however the problem is still present in release builds. Maybe it would go away entirely by increasing the DWB loop rate.

EwingKang, 2019-01-28:
I'm running the tb3 simulation (the hexagonal map), got the same issue. The sim can often finish most of the global plan, and got oscillation failure at the final orientation correction stage. The problem seems more like a bug than just a simple sensitivity adjustment in my test.

crdelsey, 2019-01-29:
@EwingKang Thanks for the feedback. I had made this one a medium priority since the robot seemed to mostly get where it needs to go, and on a failure, I could immediately retry. How much is this impacting you?

EwingKang, 2019-01-30:
@crdelsey 
I've captured a [screen recording](https://youtu.be/MrszZM0Huvs), the problem I'm seeing is listed in the video description.  
There are three main problem,  
 1. The robot is oscillating around Global path
 2. Sudden controller fail before turning to final orientation
 3. Controller seems to lag behind and drive the robot toward obstacle (and causing 1.) 
   
Since I'm running everything on the same machine, I'm trying to rule out resource starvation by running simulation on another computer. However the simulation still seems slow so far, I'm still trying.

crdelsey, 2019-01-30:
The DWB controller by default should be publishing an updated velocity every 0.1 seconds. The one in your video is only publishing every .5 to 1.0 seconds which will cause instability and oscillation.

Did you build the navigation stack yourself or are you running the crystal release binaries? If you built it your self, did you build a release version? ie. `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release` If not, try rebuilding as a release build. It gets much better performance that way.

mkhansen-intel, 2019-01-31:
@EwingKang - thanks for the input. Your video does seem to show a performance problem. I don't see this when running on my Skylake workstation even with Gazebo running on the same system. So please do try the release build as @crdelsey  mentioned above, that might be the cause of the problem. Please let us know if that helps.

EwingKang, 2019-01-31:
@crdelsey  I did build the latest release (master), will try the crystal release as you said.  

Latest update: I've ran the simulation on one machine and Nav2 the other. The oscillation fail right before reaching the goal problem still exists. 
https://youtu.be/nd-DDscViQo

@mkhansen-intel thanks for your info, will try some other setup
The oscillation problem gets better, however the CPU usage is still very high at 100%, and focused on a single core. Sometimes it would get some random deviation and failure, but I think this (CPU & random failure) probably belongs to another another issue ticket.

mkhansen-intel, 2019-01-31:
Did you build with the build flag he mentioned:
`DCMAKE_BUILD_TYPE=Release`

EwingKang, 2019-02-11:
Sorry for the late replay, it's Chinese New Year vacation here.
I've tested build with the flag you've mentioned, here's the result.
https://youtu.be/9Ji5R6z1fcQ
The oscillation detection problem still presents.
Initial controller rate seems good, however, another interesting phenomenon is that the CPU usage grows and eventually effect the execution rate as I set more goals. Maybe there's a memory leak?
I'll open another issue for that.

crdelsey, 2019-02-11:
@EwingKang Thanks for testing this. It looks like the robot now follows the path smoothly if it starts. You now only have the oscillation detection problem at the start and end of the path.

I'd appreciate it if you'd open an issue for the CPU usage increasing over time. I thought I'd seen that but haven't had a chance to really understand the problem yet.

In addition, in your video you sometimes got an error in the middle of the path and the robot would stop. That problem is noted in #514.

crdelsey, 2019-02-28:
Looks like the oscillation detection is due to oscillation in the Y axis. In gazebo right now we have a very noisy IMU. When the robot is going slow it checks Y and theta for oscillations, due to the noise, Y has oscillated.

Since we have no control over Y velocity, every proposed trajectory contains the same Y velocity and they all fail due to oscillation.

Seems like we shouldn't check for oscillation in an axis we don't control.

PR #588 (2019-03-12): Fix oscillation problem and tune DWB critics
## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | fixes #520, fixes #570  |
| Primary OS tested on | Ubuntu |
| Robotic platform tested on |Turtlebot 3 in gazebo |

---

## Description of contribution in a few bullet points

* Disables oscillation detection on the Y-axis if the robot is not capable of Y-axis motion (eg. a differential drive robot) . See #520 for reasoning.
* Rough tuning of the DWB critics to get reliable motion. I upped the obstacle avoidance critic and the rotate to goal  [... cut]
closed 2019-03-12 by crdelsey