ros-navigation/navigation2 #528: Make A* the default global planner

crdelsey, 2019-01-22:
## Feature request

#### Feature description
A* is the most commonly used global planner for navigation tasks. It should be the default used by the navigation 2 stack

#### Implementation considerations
This should just be a matter of changing default parameters in the global planner. However, testing is needed to ensure it is reliable, and that better performance is achieved.

EwingKang, 2019-01-31:
I've tried to set the A* as the default global planner in the launch.py, however, the trajectory seems to be quite rugged compare to the relatively smooth planning of the default (dijkstra?). Sometimes it would even show some weird triangle along a straight path, that is clearly not the best path. 
Not sure if this is a bug or I misjudge the performance of A*. Unfortunately I don't have any screenshots. Will post some when I'm available.

mkhansen-intel, 2019-02-15:
@orduno - have you tried the A* algorithm and have you seen this?

crdelsey, 2019-02-20:
I tried the A* planner and during my tests, the resultant path seemed acceptable. It might have had a bit of extra wobble to it. I'll have to do more testing.

I also ran 100 iterations of the system test and it passed as well as dijkstra.

Given that, I figured I'd go ahead and enable it by default on master. That way we get more eyes on it. PR #571.

I'll continue testing in the meantime.

SteveMacenski, 2019-02-20:
phew, I did that refactor and worried I messed something up.

As a side note: the navfn planner in ROS1 *never* actually had the A* implementation of navfn exposed. For A* we all generally used the `global_planner` which by itself has a number of problems that David could better explain than me. 

If you see issues in A* in navfn, you might not be crazy, it might just have issues that we can work out. If we determine its an issue with the implementation, I'm more than willing to produce another implementation. Motion planning is my jam

mkhansen-intel, 2019-02-20:
Good job @crdelsey, and thanks @SteveMacenski for the offer to help resolve issues with A*

crdelsey, 2019-02-20:
> For A* we all generally used the global_planner

Sounds like we need to do some comparisons between navfn in A* mode and the ROS1 `global_planner` to ensure we are achieving comparable performance then.

SteveMacenski, 2019-02-20:
Perhaps, you'd have to ping @DLu on how similar they are since `global_planner` was greatly his work. He might have some insights on it

DLu, 2019-02-20:
[Main Source](https://github.com/locusrobotics/robot_navigation/tree/master/dlux_plugins)

There's two main parts of the global planning algorithm. A*/Dijkstra are options for what I call the Potential Calculation, and then there's different options for the traceback. 

[image]

In the above diagram, the blue path traces back a path using each cell's four neighbors. The red path uses the eight neighbors, and the green uses a fancy gradient descent algorithm (which is what navfn did by default)

If you use Dijkstra (as in the above diagram), you expand a lot of "extra" potential, but that means there's lots of numbers to calculate the gradient off of. 

Depending on which heuristic you use with A*, you can get a variety of different (technically valid) paths that may seem "rugged" because there isn't as much potential to calculate a gradient with. Example: 
[image]

I don't have hard numbers to back up these assertions at the moment, but in general my experience has been
 * When doing planning on "typical" size maps, Dijkstra is often faster than A*, given that maintaining a priority queue in A* can be quite costly. 
 * There is a complicated optimized version of Dijkstra in navfn that I never fully understood. When I ported it to `dlux`, the result was slower than my straight forward implementation of Dijkstra, but its entirely possible that the subtleties of that optimization were lost on me / in my translation. 
 * It's possible that the A* in navfn was more efficient than my implementation, but I never tested it.

crdelsey, 2019-02-21:
@DLu Thanks for the background. The explanation for the ruggedness of the path using A* likely explains what I was seeing.

EwingKang, 2019-02-21:
Those orange spikes in the second picture exactly resembles what I'm seeing in the A*!  Thank @DLu for the explanation.

orduno, 2019-02-21:
Sounds like the best option, for now, is to keep Dijkstra as default. We would need to test with "large" size maps to evaluate performance.

We should also try the `dlux_planner` with nav2! @DLu is it in your plans to port `dlux` to ROS2?

crdelsey, 2019-03-25:
Blocked until we have a good way of evaluating the results

DLu, 2019-06-18:
Not that there's anything to report, but the issue will be tracked here: https://github.com/locusrobotics/robot_navigation/issues/45

crdelsey, 2019-10-01:
Closing since this isn't addressing a specific need. We can reinvestigate when performance testing shows an actual problem

closed 2019-10-01 by crdelsey

SteveMacenski, 2019-10-02:
@crdelsey well more than anything else, has anyone set the use A* parameter and shown that it actually _works_ ?

crdelsey, 2019-10-02:
In the system test, test_bt_navigator has it turned on while test_bt_navigator_dijkstra has it turned off

SteveMacenski, 2019-10-02:
Awesome

jihyunl2, 2024-02-20:
@orduno Are there any updates on "large" map performance with Navfn Dijkstra planner? We are using this as our default planner, and with a 1772x1077 px map, we're having trouble getting even 5Hz frequency with ComputePathThroughPoses. We want to check whether it's a computing power issue or something else isn't right.