ros-navigation/navigation2 #533: Check for collision before executing the spin primitive

orduno, 2019-01-24:
The spin primitive from `nav2_motion_primitives` should check, based on the robot footprint, if the path is clear before execution.

orduno, 2019-05-09:
@bpwilcox since you are working on the collision checker, I'll reassign it to you and move it to "in-progress".

SteveMacenski, 2019-05-15:
@bpwilcox I started to look at this. How Navigation did it was through the [costmap model](https://github.com/ros-planning/navigation/blob/melodic-devel/base_local_planner/src/costmap_model.cpp) which requires access to the costmap object. This wasnt a problem with the plugin infrastructure since it was in the same node. I'm curious if you're planning on retooling it to work with the occupancy grid message.

Let me know if I can help in some way.

bpwilcox, 2019-05-15:
@SteveMacenski I am soon to submit a PR to address this. Yes, I am using the same costmap model, so I am subscribing to the costmap topic (actually I am now publishing a nav2_msgs Costmap msg) and creating the costmap object for use in collision checking.

crdelsey, 2019-07-03:
Done now

closed 2019-07-03 by crdelsey