ros-navigation/navigation2 #540: Considerable delay from the time the goal provided to start of planning

orduno, 2019-01-28:
## Bug report

**Required Info:**

- Operating System: Ubuntu 16.04
- Installation type: from source
- Version or commit hash: head of master
- DDS implementation: Fast-RTPS

#### Steps to reproduce issue
- Launch gazebo, rviz and navstack (with bt_navigator)
- Provide robot initial pose in rviz
- Provide goal pose in rviz

#### Expected behavior
Once goal pose is provided, planning should happen almost immediately. 

#### Actual behavior
There is a considerable delay between the time the goal pose is sent and the time the `navfn_planner` starts planning.

#### Additional information
This could be related to the creation of the behavior tree since the delay is not noticable if using `nav2_simple_navigator`.

crdelsey, 2019-01-28:
May be related to #446

mkhansen-intel, 2019-02-15:
@orduno - do we have root cause on this issue yet?

orduno, 2019-05-17:
Let's test it again once we get lifecycle in.

crdelsey, 2019-07-03:
@orduno Do you still see this? I think it is gone now.

SteveMacenski, 2019-09-04:
I dont think this has been an issue within recent memory (and my memory is bad)

closed 2019-09-04 by SteveMacenski