ros-navigation/navigation2 #541: Replace nav2 tasks with ROS 2 actions

crdelsey, 2019-01-28:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2019-05-15:
I'm looking at this right now and wanted to get some feedback before going to far with it. 

I'm of the opinion that `nav2_tasks` with the Tasks are awfully confusing and breaking encapsulation of ideas. From reading the ROS2 actions demos, they look very familiar to ROS1 types (without preemption :face_with_head_bandage: ). Rather than wrapping them in a `task/util` I think the actions should be directly used in the nodes, or as a header-only wrapper in the package where they belong that the implementation inherits from. 

This buys us:
- once again encapsulating ideas
- all the stuff you need to understand a node in a package
- *greatly* simplifying the `task` package, to a point we could probably just move those elements left into `common/util` such that we only have 1-2 total helper packages
-  actions are a base type in ROS, and given that all you really need to do is set some params and register some callbacks, it doesn't really make alot of sense to me to wrap them, but we could still have a general wrapper.

crdelsey, 2019-05-16:
A bunch of changes have already been made to make the switch to actions and get rid of nav2_tasks. Unfortunately they are mixed in with the lifecycle changes right now.

I think all that is left of nav2_tasks is a wrapper around the action server to implement the preemption mechanism.

closed 2019-06-10 by crdelsey