ros-navigation/navigation2 #548: Maps from sub-map sources

gbiggs, 2019-01-31:
Relates to #392.

Enhance the map server to allow loading multiple smaller maps and serving them as a single larger map.

An example use case is an occupancy grid for a multi-building complex (like a hospital) where each building (or even each floor?) is maintained as an individual map (image) file, and these are loaded at runtime and served as a single occupancy grid map.

A possible extension or related feature is to allow requesting bits of a map on demand for navigating in extremely large areas.

SteveMacenski, 2020-10-06:
This is still in scope for how to:
- deal with very very large maps we need to shard for memory conservation
- support multiple map sources for multi-story buildings or several maps for different areas.

SteveMacenski, 2022-06-03:
https://github.com/ros-planning/navigation2/issues/1226 supersedes

closed 2022-06-03 by SteveMacenski