ros-navigation/navigation2 #554: Deterministic bringup and shutdown

mjeronimo, 2019-02-04:
## Feature request

#### Feature description
Increase the reliability and determinism of the Navigation2 bring-up and shutdown sequences. 

#### Implementation 

- [ ]  Create a base class that derives from the ROS2 LifecycleNode (to allow for adding our own node services and place missing functionality)

- [ ] Update all of the top-level Navigation2 ROS2 nodes to derive from the new Lifecycle Node class

- [ ] Update support classes to accept Lifecycle Nodes (TaskClient, TaskServer, Robot, etc.)

- [ ] Submit PRs to the ROS2 code to integrate changes required for Lifecycle Nodes (add missing functionality to LifecycleNode, MessageFilters, tf2_ros::Buffer, etc). 

- [ ] Create launch script(s) that interfaces to the Lifecycle nodes to bring up and shut down the system

- [ ] Optimize the usage of the middleware (RTPS messaging) to ensure reliability

- [ ] Test the resulting solution and assess the result

mkhansen-intel, 2019-02-08:
@mjeronimo - I'm glad you opened this issue. I have some thoughts on how to implement lifecycle nodes. First, I'm trying to be considerate of the fact that @SteveMacenski mentioned that some users may want to use nodes a-la-carte, like AMCL for example. So if that is the case these are my thoughts, for discussion purposes.

Can we refactor our code in such a way that we can have regular nodes and lifecycle nodes with minimal code duplication? In other words, have a amcl base class, an amcl 'regular' node which contains that base class, and an amcl 'lifecycle' node which also contains that class. For the amcl 'lifecycle' node, it will contain all the extra needed functions to be a lifecycle node, go through the state transitions, etc.  It would be built in the CMakeLists as a separate executable, and would be launched using a different launch file. This will add lifecycle nodes without forcing everyone to use them, but it does require more refactoring during implementation. 

It basically would allow us to support both nodes and lifecycle nodes. Let me know what you think.

SteveMacenski, 2019-02-08:
That sounds like a way to handle it that would work. I'm not opposed to lifecycle nodes as the default use for AMCL, costmap server, etc - its only that no node/server should strictly require the existence of another to run, that generally breaks the encapsulation of ROS nodes and makes the navigation stack monolithic "take it all or leave it".

Some users may just take AMCL, another common thing is to completely throw out AMCL in favor of a life long mapping solution, visual localization, or any other package for local landmark identification. Similar with your planners and controllers, for a Roomba, clearly A* isnt what you want, or if you're a omni-drive robot, DWB isn't probably the right solution for you.

The navigation stack making use of servers and plugins have been great to let anyone swap out one item for another if you just fill out the API or broadcast the same transforms. Moving away from plugin lib where a library like `move_base` would itself load the costmaps, planners, controllers, etc into independently spinning servers adds the additional headache of "is this running", but it shouldn't not allow the system to bringup (or kill other things) if one or more elements don't appear or disappear and come back up (ie if my planner server goes down, that shouldn't stop navigator from trying to do its thing. Maybe it's just respawning or its an intentional bring down to bring up a different replacement).

mkhansen-intel, 2019-02-09:
> Rather making a bringup process who's responsible for doing the maintenance or assurances makes way more sense to me and that can just be parameterized as to who to expect to come up before giving everyone registered the thumbs up.

I'm not sure I follow. Can you explain what you mean by this?

SteveMacenski, 2019-02-09:
Sure,
Instead of AMCL waiting on say, the global planner before coming up (I dont know what it would, but for arguments' sake) and therefore tying AMCL package to the global planner package for which potentially a user doesn't want one or the other, we have a separate navigation2_bringup package process who's responsible for the assurances and checks that the lifecycle nodes provide.

You could make a yaml file with the expected servers/nodes/names to bringup before shooting all the processes the thumbs up that everyone was brought up successfully and then you can now proceed to operate. Centralizing it into 1 file lets someone say "I want to remove DWB server" do a 1 liner change in their launch file to the new package and a 1 liner change to a yaml file to change DWB server to TEB server or something similar, and not hardcode into another package the need for a specific package, broadcasting specific things to exist. Modifying AMCL code to change global planners would be silly and require a massive fork to swap out components vs getting AMCL + others from apt get and updating a yaml file for the nodes used in their configuration from the general stack and adding in the names of their proprietary entries

mkhansen-intel, 2019-02-14:
I think we're not too far off in our thinking. I am thinking there's two aspects to this:
1. refactoring functionality so that each module (say AMCL) has all of its 'real' functionality in a base class. Then having 2 different wrappers for that class: amcl_node and amcl_lifecycle. amcl_node will behave much the same as the node today does. amcl_lifecycle however will be a lifecycle node and require a lifecycle node manager to bring it to the active state. 
2.. The node manager can be parameterized using YAML files, such that it is not hard-coded which nodes it brings up in which order. I think this meets your request above. This way nodes could be added / removed / re-ordered without needing to rebuild the lifecycle node manager.

@mjeronimo - care to comment on this?

mjeronimo, 2019-02-14:
@mkhansen-intel @SteveMacenski 
Hey guys, sorry for the delayed response. Just catching up on this thread...

First, having a separate base class that contains the functionality and is not ROS-specific is, IMO, always a good thing. This would take a lot of refactoring for some of the current modules (AMCL, DWB, in particular). Given my priorities for the year, this isn't something I'm likely to be able to do myself, but I do support it.  

As far as building both non-lifecycle nodes and lifecycle nodes, I don't think this is worth the trouble. Using a lifecycle node simple requires that you transition it through a couple states (Configure and Activate). This is a few lines of code in a launch file (LifecycleNodes are supported in launch). However, if the refactoring was done, this is a possibility if adding the launch lines was not acceptable.

Finally, the concern about using one or more of the nodes in a separate project: there are no additional dependencies introduced; one can simply grab and use the node as before. The only additional requirement is configuring it and activating it.

mkhansen-intel, 2019-02-14:
@mjeronimo -- can you point us to an example launch file for doing the state transition?

crdelsey, 2019-02-15:
FYI: In #568, I've added a function that can be used to bringup a lifecycle node. It makes it a one-liner in code. I added a command line utility as well just in case it is useful for backward compatibility. (It could be used in a launch file if the existing launch mechanism is too unwieldy). You'd just do `ros2 run nav2_util lifecycle_bringup <node name> ...`

closed 2019-06-10 by crdelsey