ros-navigation/navigation2 #564: Growing CPU usage over time

EwingKang, 2019-02-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Build from source, `--cmake-args -DCMAKE_BUILD_TYPE=Release`
- Version or commit hash:
  - 84b85da5f523cd26f4c23b88d6f7424c2e6422ad
- DDS implementation:
  - Default
- Client library (if applicable):
  - 
- Hardware
  - Navigation: i7-6700TE 
  - Simulation: i7-7700 
- Network
  - Closed ethernet LAN, 2 ROS2 machines only

#### Steps to reproduce issue

Launch everything with steps described in the [bringup's README.md](https://github.com/ros-planning/navigation2/blob/master/nav2_bringup/README.md#launch-navigation2-in-simulation-with-gazebo-first-time-users)
Repeat goal setting cycle

#### Expected behavior
Constant computer resource requirement.
#### Actual behavior
CPU usage seems to be growing over time.
#### Additional information
Video: https://youtu.be/9Ji5R6z1fcQ

mkhansen-intel, 2019-02-15:
@EwingKang - thank you for filing the issue. Can you also send us the output of 'top' both before the first run (in the idle state waiting for the first goal pose), and after the last run? We can also try to reproduce this by running repeatedly on our side.

EwingKang, 2019-02-20:
Hello @mkhansen-intel, I've recorded a screen capture with `top` monitor:
https://youtu.be/ek63OCAlMUg

And here is the after-shot of the screen. The process is dwb_controller
[image]

mkhansen-intel, 2019-02-20:
Thanks for the video and screenshot. @orduno was going to see if he can reproduce this.

I'm not sure I understand the issue here. I see dwb_controller does increase CPU usage to 100% when the robot is actually moving, which is somewhat to be expected as dwb_controller is running in a control loop updating about 10x/second. It seems that each time the goal is reached however, the dwb loop exits and the CPU amount consumed drops back down to about 10%. The very last screen before you exited, it was at 12%, but since we don't know if it had fully gone idle, I'm not sure if that was still going down or not. 

Is there something else that I'm not seeing on video? If so, please explain.

crdelsey, 2019-02-20:
Here are the things I noticed:
1. The first run of DWB only consumed about 40% CPU while active. By the end it was consuming around 110% while active. It increased quickly during the first few runs, more slowly after that.
2. DWB consumed about 1.1% memory at the first run. It was consuming 4.4% by the end. It was steadily increasing during the whole video.

EwingKang, 2019-02-21:
@crdelsey pretty much speaks all what I'm noticing. Another thing is that the controller rate is decreasing over time. Since the CPU usage for controller is maxed out at around 100% (because it's a sequential procedure I presume?), my explanation toward decreasing rate of the controller is that the computational requirement is growing.

I have no hard evidence, but a very short prediction path like the default setting with only 10Hz should uses much less CPU. My guesses is based on my experience with the ROS1 nodes. When I'm using the same hardware rig, one particular case was I can run ROS1 TEB planner at 10Hz, with a prediction length of 4m&(0.1m/pix), and with global planner re-planning at 0.4Hz at the same time. TEB utilizes a nonlinear optimization solver on a car-like vehicle (constrained problem), which should demand much more calculations than DWA. So the 40% CPU usage in the beginning seems closer to my expectation. Again, these are only my instincts without any evidence.

orduno, 2019-02-21:
@EwingKang I got similar results for CPU and Memory (RSS) utilization:

| process | initial idle | 1st | 2nd | 3rd | (after 8min)|
|---|---|---|---|---|---|
| dwb | ~9.5%, 86M | ~80%, 101M | ~94%, 113M | ~103%, 124M |106% 345M |

The rest of the navigation processes (navfn_planner, amcl, etc) seem constant over time.

Other two processes relevant are related to gazebo, but those are constant.

| process | initial | end |
| --- | --- | --- |
| gzserver | ~38%, 271M | ~38%, 271M |
| gzclient | ~13%, 334M | ~12%, 334M|

There's a [video](https://youtu.be/b25hH63JhFk) from my test.

We will need to dig deeper into DWB.

orduno, 2019-02-21:
Looks like the text in the youtube video is hard to read -- not sure what happened there.

Here are some screen captures.

Idle:

[image]

1st:
 
[image]

after 8 min:

[image]

crdelsey, 2019-02-21:
I think the starting point would be to run dwb under `valgrind`. That will likely show the memory leak fairly easily. Maybe the increasing CPU usage is related to the memory leak.

moriarty, 2019-02-22:
I haven’t been trying any of the ROS2 Navigation2, but just thought I’d share a GUI tool which sort of wraps ‘perf’ for doing sample based performance analysis:

https://github.com/KDAB/hotspot

It’s not an alternative to valgrind, just an additional tool, which is usually very easy to setup and doesn’t slow things down like valgrind which can sometimes hide things.

We’ve used it to know where to target effort in improving performance.

orduno, 2019-02-22:
I tested #577 with good results:

| process | initial idle | 1st | 2nd | (after quite a few runs)|
|---|---|---|---|---|
| dwb | ~8.5%, 82M | ~79%, 93M | ~76%, 93M | ~77%, 124M |

I also noticed that oscillations reduced - #520.

PR #577 (2019-02-26): Fix DWB memory leak
## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | fixes #564  |
| Primary OS tested on | Ubuntu |
| Robotic platform tested on | TB3 on Gazebo |

---

## Description of contribution in a few bullet points

* Moved the DWB initialize call from the Execute method to the constructor.
* I ran Valgrind and saw a bunch of leaks related to loading the plugins in DWB and Costmap. By moving this code to the constructor, it only runs once. This makes more sense logically and makes DWB a bit faster to start moving.
closed 2019-02-26 by crdelsey