ros-navigation/navigation2 #565: Design the World Model

orduno, 2019-02-13:
# Objectives
Design a world / environmental model for 2D navigation. 

Drivers:
- Consolidate
- Parts of world model are scattered.
- World model clients tend to have their own world representation. Leading to code replication.

# Proposal

## Maps & Map Server

Map representations to support:
- Metric (2D Grid)
  - Occupancy, Cost maps
- Topological
  - Multiple graphs for a world
  - Robot can travel between nodes (assumption)
- K-d Tree (Quad/Oct trees)
- Hybrid

Road network.

Open for discussion:
- Transformation of the map into a representation useful for planning. Issue: Planner-dependent.

## Clients
Let's consider the consumers of the information contained in the world model.

Clients operate on different length-scales and use different levels of the world model.

### (Global) Path Planning
Can operate on a road network, topology map, or global map (sub-sampled occupancy grid or k-d tree).
Coupled with the map reprentation being used.

Possible types / paradigms:
- Optimal Control (LQR/G, MPC)
- Potential Field
- Cell Decomposition (This is more a graph construction technique, that then uses a graph search algo for planning. We can considered this a feature of the World Model instead)
- Graph Search (A*, D*, E* and variants)
- Sampling-based
  - Deterministic (Hybrid-A*, Lattice)
  - Random (PRM, RRT and variants)

- 

### (Local) Obstacle Avoidance and Control
Operates on a high-resolution local map representation of the world like an occupancy grid.

Planner-dependent

Possible types:
- Optimal Control (LQR/G, MPC)
- RRT

### Motion Primitives & Recovery

# Transition

# Next Steps

SteveMacenski, 2019-02-18:
On a high level I don't see anything here that I see as a show stopper, but it also seems like this design is trying to bite out alot of things to be generally and extendable to everybody. I might think about reducing scope to get something out there that works and is relatable to existing technologies to build off of.

Only bit of input I'd add is that in the "core representation" I believe that there is no single correct way of representing the space (costmap, grid map, traversability map, etc) and should instead be a combination of all of them - such that the costmap is really a vector of of these items stacked on top of each other. Different applications can utilize different representations as they like but the data filled in populates the costmap, and the traversability map, and the ... based on whatever the application warrants.

SteveMacenski, 2019-02-28:
From talking about this again this week I had another thought which I'm sure others have thought about but maybe not in words:

We would like to not have to redo different planning and controlling algorithms into different ones for different representations of costmaps. Rather than trying to make something like templating everywhere and trying to make everyone happy and readable, I'm thinking a good thought would be to have adaptors.

For example right now we're going costmap -> planner. Now we might go costmap -> adaptor -> planner, similarly with traversability map -> adaptor -> planner. This lets us generalize all the new planning and control algorithms independent of the implementation of the world model (making those by itself is hard without having to deal with lots of templates and thinking about ramifications across multiple representations).

Then we have a finite set of adaptors we need to build to represent the different models we'd like (costmaps, traversability, elevation, etc) who's job is to take the value of the neighboring cells and apply the vehicle kinematics and dynamics to give back a response "ok" "not ok" "unknown" or other options. The example adaptor for a costmap would be to change the 0-255 to those return types. Elevation map would apply the vehicle dynamics or maximum gradients to return the same type of information. 

Now all planners or controller that work for 1 will work for all as long as they can run with   "ok" "not ok" "unknown" or other options and be extendable easily for sampling based planners by creating methods in the adaptor to get the value at a certain location in global frame coordinates which the adaptor could project into a costmap cell, voxel grid pose, or elevation gradient.

crdelsey, 2019-03-07:
@SteveMacenski 
> adaptors ... who's job is to take the value of the neighboring cells and apply the vehicle kinematics and dynamics to give back a response "ok" "not ok" "unknown"

Are you envisioning the output of the adaptor in this example as a sort of simplified costmap where each cell is filled with a ok, not ok, or unknown value. Or are you thinking more that the client feeds a trajectory to the adaptor and the adaptor returns the evaluation of the trajectory as a ok, not ok, unknown value?

This pushes the knowledge of the vehicle dynamics to the world model (adaptors) instead of the algorithms. How confident are you that this is the right place for that knowledge?

I had thought of having input and output adaptors as well, but I was thinking it could result N^2 adaptors since we theoretically could want to convert from any representation to any other. I expect it wouldn't be so bad in practice since many conversions are probably not useful.

Based on your feedback, I was imagining something like this below. We have a collection of representations. Adaptors to convert incoming data to those representations as needed. Clients that either get data directly from the representation they need or from an adaptor if there is a mismatch in the representation provided and the type they need.

[image]

SteveMacenski, 2019-03-07:
Thanks for making that diagram, I'm certainly a pretty lazy guy when it comes to visualizations :) 

First off, I'm not sure why you have a separate pipeline for laser scans, that defeats the purpose of having a general purpose world model to take in arbitrary sensors to generate a view of the world. I'd recommend completely scrapping that, I don't see anything special about a laser scanner requiring that type of pipeline within the world model. If you'd like to use it as a safety sensor with zones, that should be up stream of this since that's not generalized and many robots today don't use them anymore or have a different sensor suite. 

What I think should be happening is a similar way of how it's done today. There's a set of optional plugins to buffer arbitrary sensor information  (scans, images, depth maps, radar, sonar, etc) when does stuff of interest for that sensor, and inserts it into the map. I'm not thinking greatly into how to generalize those plugins for different plugins. In practice, I dont think you'll find a way to generalize that. A costmap is binning a depth map for collision avoidance, while an elevation map will use them over time to generate an ellipsoid curve or something are very different operations on the same sensor data. Those plugins for buffering and inserting data into their representation are probably representation specific. 

Looking on the other side however, we have our [representation A] which needs to be utilized by the local/global planners to navigate. In the case of the elevation map, the Z coordinate becomes meaningful so we can't just talk about 2D X-Y coordinates anymore. The planner or controller will say "give me the neighbors" and its then up to the adaptor to say "I'm an elevation map, therefore my neighbors are in a 3D 8-directional curves" or "I'm a costmap, I just need to give the cells on either side of me", and return that information to the planner or controller to do their will with an ask for more things as needed. 

This lets us do 2D or 3D representations but allow the same algorithms for local and global planning to operate (and moreover, would work for drones as well). When the adaptors return their neighbors, its up to the adaptor's knowledge of the robot dynamics and mechanics to assign them some set of finite states that can be generalized across all representations. That might include OK, not OK, unknown, not recommended, etc but as long as all of the algorithms are built to work with those same finite states, then whatever the representation is, it'll work. And the adaptors can use the Robot class to get those relevant dynamics. For a diff drive robot, its just all OK like in navigation1, but for an ackermann car or an elevation map on a legged robot, that may not be valid

orduno, 2019-03-08:
I think what you are refering to as adaptor is what I have as a plugin below. We can later discuss the best pattern but the idea is the same, this block's goal is to translate the information available on the Core Representation and express it in something the client needs.

[image]

SteveMacenski, 2019-03-08:
Yes that's what I'm thinking, with the only exception that the "Sensing and Perception" isn't 1 plugin, but also a series of plugins, but I think that may have just been represented that way for brevity. 

Each representation will have an associated plugin that converts its values into the limited-enum types and is responsible for answering the query from the controller/planner "give me the neighbors" and "give me the value at (x,y,z)".

crdelsey, 2019-03-09:
>  I'm not sure why you have a separate pipeline for laser scans

It was meant to represent a possibility. If a sensor could output data directly in representation format, it could talk directly to the model. But that's a stupid idea in context. I was playing with the idea that everything could be a ROS node; each representation could be a node and each adaptor as well.

> I think what you are refering to as adaptor is what I have as a plugin below.

Do we want to be able to chain adaptors/plugins? As in, there is a plugin that provides data from the representation, but there is a second plugin that grabs the output of the first plugin and provides it in a different way

> Each representation will have an associated plugin that converts its values into the limited-enum types and is responsible for answering the query from the controller/planner "give me the neighbors" and "give me the value at (x,y,z)".

So we'd need to figure out the queries that can be used by many algorithms. We'd then end up with an output plugin per class of algorithm, where a class could be graph search algorithms like A*, Dijkstra, D* etc.

SteveMacenski, 2019-03-09:
Well at the end of the day all the graph search algorithms are going to ask for neighbors, the sampling based planners will ask for the result at a certain position, I'm not _totally_ certain what optimization based planners will ask for, but that's overkill for 2D navigation as far as I know. If not, we can find out what thematically they ask for. 

I think plugin API implementing neighbors, position in global frame (then it should internally find the value of its representation i.e. costmap would look for X-Y cell index and then the location in the array), and random sample to start with. We can always extend it if this is the way you also think makes sense. It was just a suggestion

SteveMacenski, 2019-04-06:
(More thoughts) 
Grid Maps exists and does a number of things. I haven't done too much work with it myself but from what I've been reading it looks like everything costmap_2d does, this does as well. We might want to consider that as an option to completely replace costmap 2d for a ground up rebuild. 
Link for reference: https://github.com/ANYbotics/grid_map

orduno, 2019-10-04:
Closing the issue, it seems we're sticking with the current costmap design for now.

closed 2019-10-04 by orduno