ros-navigation/navigation2 #572: Compare navfn A* planner with ROS 1 global planner

crdelsey, 2019-02-20:
## Description

The ROS 1 `global_planner` was the more widely used implementation of A*. We need to compare the results of the `navfn` planner in A* mode with `global_planner` to ensure we are comparable.

By performance, I mean quality of the resulting path and CPU utilization.

orduno, 2019-02-20:
When I was looking into which global planner to port (for use in navigation2) I tested the `global_planner` in ROS1 with Turtlebot3 and got some weird results, hence the decision to port `navfn` instead.

The [dlux global planner](https://github.com/locusrobotics/robot_navigation/tree/master/dlux_global_planner) might be a better candidate.

SteveMacenski, 2019-02-21:
No arguments here, I'd love to move on to use code being actively supported. Many of the pieces of the navigation stack were supported/maintained by people and companies no longer using that tool.

closed 2019-10-01 by crdelsey