ros-navigation/navigation2 #573: Navfn planner might be backtracing the path twice.

orduno, 2019-02-20:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2019-05-11:
@orduno can you elaborate on this? I can probably help out if we think this is an issue.

orduno, 2019-05-16:
@SteveMacenski I've updated the issue description with more information about the possible bug.

shivaang12, 2019-11-13:
@orduno is this still an issue? I can help if it is.

SteveMacenski, 2019-11-13:
This is still an issue, we're not 100% sure that its happening but from glancing over again, I'm about 80% sure it is. 

Feel free to submit a PR

closed 2019-11-18 by SteveMacenski