ros-navigation/navigation2 #575: Remove `stop` motion primitive, make a method on the robot class

orduno, 2019-02-21:


SteveMacenski, 2019-05-11:
@orduno given the robot class is now a util, do you think that should still apply? It could simply use the `PublishVelocity` util with an empty twist message, but I could wrap that if you like.

Else maybe this ticket is closed. Let me know :)

crdelsey, 2019-05-16:
@SteveMacenski 
> but I could wrap that if you like

I like the idea of wrapping the stop command for code readability, but I think we can close this issue when your PR gets merged.

orduno, 2019-05-17:
Closing this issue as #725 removes the `stop` motion primitive.

Basically, with the new ROS2 actions, the dwb controller will send a stop command whenever its action command is canceled. This occurs when the followPath bt action node is halted.

PR #704 (2019-07-19): Robot class to util

---

## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | https://github.com/ros-planning/navigation2/issues/692 |
| Primary OS tested on | Ubuntu |
| Robotic platform tested on | Simulation |

---

## Description of contribution in a few bullet points

* Remove the robot class 
* Replace with 2 utils: VelocityPublisher to wrap the cmd_vel topics for the recoveries, tests, and DWB planner and the RobotStateHelper which currently contains helper methods for pose and odometry information about the robot.

---

fixes #5 [... cut]
closed 2019-05-17 by orduno