ros-navigation/navigation2 #580: Enable resetting of the static layer

orduno, 2019-02-25:
Calling `StaticLayer::reset()` after the layer is intialized will result on an ROS error: Node has already been added to an executor.

https://github.com/ros-planning/navigation2/blob/1c1f4cd71ee55c13ac2864d4a6437bc4f85d87ed/nav2_costmap_2d/plugins/static_layer.cpp#L267

This error is coming from here:

https://github.com/ros-planning/navigation2/blob/1c1f4cd71ee55c13ac2864d4a6437bc4f85d87ed/nav2_costmap_2d/plugins/static_layer.cpp#L101

The `onInitialize()` method is normally called from the costmap constructor before the node is spun from main to wait for a map. When `reset()` is called, the costmap is already spinning.

We need to separate the initialization from configuration and resetting.

The solution to this is probably considered within #554.

crdelsey, 2019-03-04:
Most likely this goes away with the transition to lifecycle nodes

crdelsey, 2019-07-01:
@orduno to verify if this still is a problem.

closed 2019-10-07 by orduno