ros-navigation/navigation2 #593: Add retry capability to ServiceClient

crdelsey, 2019-03-06:
There are other places in the code besides life-cycle that could benefit from this retry method.  For example for Auto Localization currently I'm using the BT retry node for this purpose but we might need a simple retry method for other tasks.

_Originally posted by @mhpanah in https://github.com/ros-planning/navigation2/pull/568_

mkhansen-intel, 2019-03-07:
Are you saying every ServiceClient in the code base or is there a particular instance that needs this fix?

crdelsey, 2019-03-07:
I'm proposing we add the retry capability to the base class and then make use of it in every instance in the code base, unless it doesn't make sense in that specific instance.

mjeronimo, 2019-03-07:
While this could be a pragmatic work-around to a problem we're seeing, I think that reliability should be the responsibility of the DDS/RTPS layer. I also think there are potential semantic problems with the proposed approach. For example, Actions are essentially a collection of services (and topics). What if we invoke an Action, like NavigateToPose, and don't receive the response, so we invoke it again. On the receiving end, it may have been received and started a long-running task. We would need to ensure that all Services/Actions can be pre-empted with a new request. However, it's not clear to me that the new Action interface implies that a service must support preemption: https://github.com/ros2/rclcpp/blob/master/rclcpp_action/include/rclcpp_action/client.hpp

IMO, to implement services, it should be over a reliable transport.  If we're not getting reliable services, we must debug that issue.

crdelsey, 2019-03-08:
> While this could be a pragmatic work-around to a problem we're seeing, I think that reliability should be the responsibility of the DDS/RTPS layer.

I agree that the underlying code needs to be fixed. Service calls are very broken right now. However, they go over the network, there is only so much DDS can do; networks are unreliable. In the end, every time we make a service call, we should time it out and take appropriate action if it takes too long. Retrying is one possible recovery action that may or may not be appropriate in all circumstances. You are right that we need to consider carefully what recovery action we choose.

Right now, at this moment in time, with the underlying middleware in the state it is, retrying a couple of times seems likely the best course of action in many cases, and putting the retry support in the `ServiceClient` is the most convenient. I'm not sure if it is the right place though. I'm open to suggestions.

My initial thinking was to add retries as an optional argument when you invoke `ServiceClient`. eg.

```
void wait_for_service(std::chrono::seconds timeout = max(), int retries = 0);
```

Then it is still up to the client whether they want to use this or not.

I don't really like this solution, but I haven't come up with a better one yet.

> We would need to ensure that all Services/Actions can be pre-empted with a new request.

Good point. I knew this was going to be bigger than it seemed at first glance.

crdelsey, 2019-03-08:
Rather than add the retry into service client, it could just be a macro or function that gets used at a higher level. eg. the client could do something like 

```
RETRY(3, std::runtime_error) {
  service_client->wait_for_service(5s);
}
```

I'm not sure if I can pull of that syntax though

crdelsey, 2019-03-08:
Oh. One other thought.

> What if we invoke an Action, like NavigateToPose, and don't receive the response

Maybe for these core actions, we should only fail it all the way up to "Mission Executor" or the equivalent. It could then try stopping the robot and then retrying. That seems like it would be a bit annoying for users of the `Send Nav Goal` button of Rviz though, so maybe handling it in Bt and Simple navigator would be best.

Again, right now, with how we are using the nav stack and the current unreliability, retrying at the lower levels as well maybe be a good thing.

mjeronimo, 2019-03-09:
There are services, like getting a map, where invoking the service again upon failed communication, would be safe. Other services, like starting an Action, could have been initiated even if the response was not received, so they'll need to be handled with care. In this case the service may or may not need to be canceled. Or, if it can be pre-empted, another service request could be safely re-sent. As long as its under the caller's control, where presumably one would understand the implications of re-invoking a service that may or may not have been successfully sent/received.

crdelsey, 2019-07-03:
Need to verify if this is still relevant.

crdelsey, 2019-10-01:
Obsolete

closed 2019-10-01 by crdelsey