ros-navigation/navigation2 #595: Nav2 should work with Cartographer, other SLAM packages

mkhansen-intel, 2019-03-07:
## Feature request
Instead of using AMCL and map_server, a user should be able to run a SLAM package such as Cartographer or gmapping to provide the map and localization info. 

#### Feature description

This is a common usage model from ROS1 when mapping. Explore algorithms and RVIZ can then be used to drive the robot during mapping.

#### Implementation considerations

Currently the robot's pose is retrieved from the amcl_pose topic which isn't published by SLAM packages such as Cartographer.

mhpanah, 2019-03-07:
We can look at the following three scenarios with Cartographer or other SLAM packages:
1.	Map of environment is unknown and user is tele-operating the robot with Cartographer to map the environment.  Once the environment is mapped, the user will use our Nav2 stack the way it is to navigate.
2.	Map of environment is unknown and the user wants to autonomously map the environment with Cartographer.  Once the environment is mapped, the user will use our Nav2 stack the way it is to navigate.
3.	Map of environment is known or partially known; user is using Cartographer to keep updating the map while navigating.
For the second and third use cases, we can utilize behavior tree to enable/disable proper components and add proper planner within our Nav2 stack that is suitable for these tasks.

mkhansen-intel, 2019-03-08:
I agree with the 3 use cases. I think we should support all 3 with both simple_navigator and bt_navigator. Otherwise we have to document exactly which ones work with each use case.

crdelsey, 2019-03-12:
Regarding @mhpanah 's points 1 to 3 above:

1. Is already working
3. We just need to change the QOS settings on the map topics to the default so we are compatible with the maps published by Cartographer
2. Involves bringing up an exploration node to drive the mission. It's probably outside of nav2

mkhansen-intel, 2019-03-12:
Use case 3 requires the change I mentioned above. amcl_pose is not published in that case, so another method is needed to check for the robot's position.

rotu, 2019-06-20:
Trying to use Nav2 with Cartographer, one issue is that the static layer costmap plugin hangs indefinitely waiting for a service (`nav_msgs/srv/GetMap`). Even though the static plugin can subscribe to the OccupancyGrid as a topic (`nav_msgs/msg/OccupancyGrid`), it needs an option to turn off the map service client.

```
[world_model-1] [INFO] [global_costmap.world_model]: Creating
[navfn_planner-2] [INFO] [navfn_planner]: Creating
[world_model-1] [INFO] [world_model]: Configuring
[world_model-1] [INFO] [global_costmap.world_model]: Configuring
[world_model-1] [INFO] [global_costmap.world_model]: Using plugin "static_layer"
[world_model-1] [INFO] [global_costmap.world_model]: StaticLayer: Requesting map from the map service
[world_model-1] [INFO] [world_model_client]: map service client: waiting for service to appear...
[world_model-1] [INFO] [world_model_client]: map service client: waiting for service to appear...
[world_model-1] [INFO] [world_model_client]: map service client: waiting for service to appear...
```

rotu, 2019-06-21:
Another issue is that Cartographer ROS only publishes transforms, not poses (nav stack expects PoseWithCovariance):

https://github.com/ros2/cartographer_ros/issues/30

crdelsey, 2019-06-24:
> Another issue is that Cartographer ROS only publishes transforms

PR #793 should address that problem. However, when I tested with that code enabled a month or two ago, both DWB and Cartographer crashed frequently, so I expect there is still more to debug.

mkhansen-intel, 2019-09-17:
This is working now, fixed by #1057

closed 2019-09-17 by mkhansen-intel

mhaboali, 2020-06-06:
> Regarding @mhpanah 's points 1 to 3 above:
> 
> 1. Is already working
> 2. We just need to change the QOS settings on the map topics to the default so we are compatible with the maps published by Cartographer
> 3. Involves bringing up an exploration node to drive the mission. It's probably outside of nav2

Hi 
Thanks for this great discussion, I'm wondering now about how can I run the scenario #2 as described here? I mean to run nav2 with cartographer without using mapserver.

Thanks!

mhaboali, 2020-06-07:
Hi @mkhansenbot 

Could you help me with running the Nav2 with Cartographer without a previously recorded map? And how can I disable the map server in this case?

You help will be appreciated!

SteveMacenski, 2020-06-08:
Please file a question on ROS Answers.