ros-navigation/navigation2 #600: Merge MissionExecutor and BtNavigator

mjeronimo, 2019-03-08:
## Feature request

#### Feature description
The MissionExecutor and BtNavigator are quite similar and could be merged into a single generic BT execution node. The purpose of the node would depend on the BT that it is executing. It could be executing a mission, navigating to a pose, or whatever, depending on the BT that it has been provided to execute. 

The work would consist of:
- Removing the nav2_mission_executor module
- Renaming nav2_bt_navigator to something like nav2_bt_executor
- Updating launch scripts
- Updating the BT README

If one wanted to have multiple nodes that each are executing a BT, one would simply instance the BtExecutor node multiple times, each with a different XML input file.

mkhansen-intel, 2019-03-08:
I like it, provided we still have an easy startup for the default navigation behavior tree (I think you implied this above, just want to make it explicit). Many end users will just use the default.

SteveMacenski, 2019-03-08:
Weren't these separate so that there might be a cloud-based mission planner/coordinator separate of the current set of goals on the robot for navigation?

I tend to agree these should probably be in the same class for use-cases I can think of but curious why the change in direction?

On the naming front, then the only "navigator" package is the simple navigator. Perhaps some naming or architecture should be reconsidered for that as well.

mkhansen-intel, 2019-03-08:
Yes, they were, and the idea of having a mission planner and higher level mission executor is still in our future goals. However, the beauty of the behavior tree is that it's defined in XML, so you could actually use the "bt_executor" as a mission executor just by changing the behavior tree XML being run. So I don't think this precludes us from doing this in the (hopefully near) future.

mjeronimo, 2019-03-09:
Yes, as Matt said it's not a change in direction, but a recognition that the same node could fufill both purposes by simply using a different BT XML input. The only difference currently is that the MissionExecutor takes the input via a topic and the BtNavigator via a parameter. If we unified it and specified the input XML via a parameter, it could be used for both purposes (and perhaps others).

SteveMacenski, 2019-03-09:
Got it. Given that the thought is to rename it to `nav2_bt_executor` and there still exists the `nav2_simple_navigator` which fulfills the same purpose, I think then the navigator needs a rename as well, or else it may look like there's only 1 navigator and the other is somehow for a different purpose. An example is if we had a package `nav2_a_star_trajectory_planner` and `nav2_dijkstras_path_planner`, sure, they're _very_ similar but the choice of words differing would be distracting from learning how everything fits together

mjeronimo, 2019-03-09:
Yeah, I can see how that wouldn't be obvious to the newcomer. Once we make the BtNavigator the default, we could drop the simple navigator completely I suppose.  It's been useful to have during the development of the BT functionality, mostly as a stable backup/reference. Otherwise, I'm open to naming suggestions that would help to clarify.

mkhansen-intel, 2019-03-11:
> The only difference currently is that the MissionExecutor takes the input via a topic and the BtNavigator via a parameter.

I think if we merge the two we need it to be possible to call it via a topic or preferrably an action, as the "Mission controller" might be a remote server, so it would send the BT XML to the bt_executor remotely, and have a way of knowing when the mission was completed.

crdelsey, 2019-03-13:
> I think then the navigator needs a rename as well, or else it may look like there's only 1 navigator

What we could do is make a `bt_engine` package for the common implementation, but keep the `bt_navigator` and `mision_executor` packages just to hold the respective behavior trees, docs, and example launch files.

crdelsey, 2019-10-28:
Closing this since it is obsolete. Mission Executor is gone now.

closed 2019-10-28 by crdelsey