ros-navigation/navigation2 #601: Release for Crystal sync on 3/14

mkhansen-intel, 2019-03-11:
## Feature request
There's a Crystal sync this week on 3/14.
https://discourse.ros.org/t/preparing-for-crystal-sync-and-patch-release-2019-03-14/8252
Let's do a release 0.1.6

mkhansen-intel, 2019-03-12:
@SteveMacenski - do you have time to do a release? I think @crdelsey has 3 bug fixes he wants to backport to crystal_devel from master first

SteveMacenski, 2019-03-12:
I was waiting on your end to say that you're happy with the state of it to release, just let me know and I can probably do it day of

mkhansen-intel, 2019-03-12:
OK let's try to do it by tomorrow (3/13)

SteveMacenski, 2019-03-12:
Is everything in on the crystal branch you want now?

crdelsey, 2019-03-12:
I just created a PR with the changes (#604). I want to make sure it passes the build farm first. I'd like to do a bit more testing on it as well.

SteveMacenski, 2019-03-12:
Ok, feel free to comment here or email me when you're all happy with it

crdelsey, 2019-03-13:
@SteveMacenski I merged the changes in. It is good to go now.

SteveMacenski, 2019-03-13:
In progress

SteveMacenski, 2019-03-13:
@crdelsey @mkhansen-intel submitted, watch the build farm for additional issues with dependencies since last sync. Here's the PR to rosdistro: https://github.com/ros/rosdistro/pull/20552

mkhansen-intel, 2019-03-14:
Looks like all is well! Thanks @SteveMacenski!

closed 2019-03-14 by mkhansen-intel