ros-navigation/navigation2 #603: Update "Ignore Y oscillation on robots not capable of Y velocity"

crdelsey, 2019-03-12:
PR #588 has some unfinished work that needs to still be done

- [ ] Ensure default critic scale parameters are being set and used correctly in the code. We shouldn't need to set them in the params.yaml file unless we want to change it from the default.
- [ ] Work with upstream DWB to ensure the Y oscillation problem is fixed there as well
- [ ] Add parameters to the oscillation critic to allow the user to choose which axes will be considered for oscillation detection
- [ ] Generalize the existing mechanism to all axes.

mkhansen-intel, 2019-04-24:
@crdelsey -Is this fixed by #674?

crdelsey, 2019-04-25:
Not entirely. #674 would fix bullets 3 and 4 above but not 1 and 2.

closed 2019-06-10 by crdelsey