ros-navigation/navigation2 #605: Update build script

owenvt, 2019-03-13:
## Bug report

**Required Info:**

- Operating System:
 Ubuntu 18.04
- Installation type:
ROS2 Crystal, from binaries
- Version or commit hash:
crystal-devel 0.1.6

#### Steps to reproduce issue

```
Follow instructions listed in BUILD.md:
https://github.com/ros-planning/navigation2/blob/master/doc/BUILD.md
```

#### Expected behavior
Package to build successfully

#### Actual behavior
Various build errors, this first of which is below:
> /root/ros2_ws/src/navigation2/nav2_util/src/node_utils.cpp: In function 'rclcpp::Node::SharedPtr nav2_util::generate_internal_node(const string&)':
> /root/ros2_ws/src/navigation2/nav2_util/src/node_utils.cpp:68:11: error: 'NodeOptions' is not a member of 'rclcpp'
>    rclcpp::NodeOptions options;
>            ^~~~~~~~~~~
> /root/ros2_ws/src/navigation2/nav2_util/src/node_utils.cpp:68:11: note: suggested alternative: 'InitOptions'
>    rclcpp::NodeOptions options;
>            ^~~~~~~~~~~
>            InitOptions
> /root/ros2_ws/src/navigation2/nav2_util/src/node_utils.cpp:69:3: error: 'options' was not declared in this scope
>    options.use_global_arguments(false);
> 
> Failed   <<< nav2_util	[ Exited with code 2 ]
> Aborted  <<< nav_2d_utils

#### Additional information

This is what I did to get the build working for me:
1.     Modify the provided installation script (https://github.com/ros-planning/navigation2/blob/crystal-devel/doc/BUILD.md) to use crystal, not bouncy, and run with --no-ros2 argument.
2.     Clone geometry2 into navstack_dependencies_ws/src/gazebo_ros_pkgs (https://github.com/ros2/geometry2/tree/crystal)
3.     Checkout the crystal versions of navigation2, BehaviorTree.cpp, gazebo_ros_pkgs, and geometry2 (this didn't work on all of them inside docker, but it was okay).
4.     Remove the gazebo_ros and gazebo_plugins folders from navstack_dependencies_ws/src/gazebo_ros_pkgs to prevent libigition protobuf compile errors.
5.     Remove the image_common folder from navstack_dependencies_ws/src/ if it exists.
6.     Clone https://github.com/ros-perception/vision_opencv into navstack_dependencies_ws/src and checkout ros2 if it doesn't exist.
7.     Navstack_dependencies_ws should now build with colcon build, build it and . install/local_setup.bash
8.     Navigation2_ws now builds as well, run colcon build and . install/local_setup.bash

crdelsey, 2019-03-14:
@owenvt That build script has definitely been neglected for a while. Thanks for the report.

SteveMacenski, 2019-10-15:
I think with the recent build updates from @mlherd we can close this

closed 2019-10-15 by SteveMacenski