ros-navigation/navigation2 #615: Running nav2_bringup_2nd_launch.py caused cartographer to stop updating the map

crdelsey, 2019-03-19:
## Bug report

**Required Info:**

- Operating System:
  -Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - aa9b45cbb0e43
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue
On the turtlebot 3, I brought up all the nodes as decribed by the TB3 manual up to running cartographer. However, I skipped the teleop node so I could run the nav stack instead.

After running nav2_bringup_2nd_launch.py, and then killing it and running teleop instead, cartographer was no longer providing map updates.

This may not be a nav stack issue, but it needs further investigation.

mlherd, 2019-08-24:
PR #1057 fixes this bug.

SteveMacenski, 2019-09-04:
2nd launch file is dead, so shall this ticket

closed 2019-09-04 by SteveMacenski