ros-navigation/navigation2 #631: Random Crawl

mhpanah, 2019-03-25:
## Feature request
Random Crawl
#### Feature description

Utilizes Reinforcement Learning to make turtlebot 3 to randomly crawl around without hitting any obstacles.

- Have already implemented this task internally. Will need to bring it out to our repo.

SteveMacenski, 2019-03-26:
Is this based on the turtlebot3 DQN tutorial? I had just run across that and seemed similar. If you hadn't seen it, another good reference implementation 

Link: http://emanual.robotis.com/docs/en/platform/turtlebot3/machine_learning/#machine-learning

mhpanah, 2019-03-26:
> Is this based on the turtlebot3 DQN tutorial? I had just run across that and seemed similar. If you hadn't seen it, another good reference implementation
> 
> Link: http://emanual.robotis.com/docs/en/platform/turtlebot3/machine_learning/#machine-learning

Yes, I've seen this and it's a great tutorial.  The random crawl is currently also using DQN, however the implementation is slightly different.

PR #683 (2019-05-10): Experimental Random Crawl task implemented with Reinforcement Learning (DQN)

---

## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | (fixes #631) |
| Primary OS tested on | (Ubuntu 18.04) |
| Robotic platform tested on | (Turtlebot3 gazebo simulation) |

---

## Description of contribution in a few bullet points
Utilizes Reinforcement Learning to make turtlebot 3 to randomly crawl around without hitting any obstacles.

## Future work that may be required in bullet points

- Improve collision detection method. Currently, the collision  [... cut]
closed 2019-05-10 by mhpanah