ros-navigation/navigation2 #636: Data race condition on variables of nav2_costmap_2d::StaticLayer

vinnamkim, 2019-03-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - latest commit

#### Steps to reproduce issue
Add these line on functions of nav2_costmap_2d, build, and run nav2_world_model
```
std::cout << "StaticLayer::incomingMap Thread : " << std::this_thread::get_id() << std::endl;
std::cout << "StaticLayer::reconfigureCB Thread : " << std::this_thread::get_id() << std::endl;
std::cout << "StaticLayer::updateBounds Thread : " << std::this_thread::get_id() << std::endl;
```

#### Actual behavior
StaticLayer::updateBounds Thread : 140293057500928
StaticLayer::incomingMap Thread : 140294026309440

The address of Map Update Thread and Incoming Map Thread are not same.

#### Feature description
Map update thread is separated from the ROS callback thread. So, It seems the variables on nav2_costmap_2d::StaticLayer are on data race condition. For example, nav2_costmap_2d::StaticLayer::has_updated_data_ can be changed by nav2_costmap_2d::StaticLayer::incomingMap, reconfigureCB, and updateBounds.

#### Implementation considerations
Use std::atomic<bool> to these variables

crdelsey, 2019-03-28:
@vinnamkim Thanks for reporting this.

SteveMacenski, 2019-05-11:
Closing from resolution in #639

closed 2019-05-11 by SteveMacenski