ros-navigation/navigation2 #656: nav2_util link error in macos

routiful, 2019-04-10:
## Bug report

**Required Info:**

- Operating System:
  - macOS High Sierra v 10.13.6
- Installation type:
  - source
- Version or commit hash:
  - master branch
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Just `colcon build`

#### Expected behavior

#### Actual behavior
```
--- stderr: nav2_util
Undefined symbols for architecture x86_64:
  "_map_calc_range", referenced from:
      nav2_util::BeamModel::sensorFunction(nav2_util::LaserData*, _pf_sample_set_t*) in beam_model.cpp.o
  "_map_update_cspace", referenced from:
      nav2_util::LikelihoodFieldModel::LikelihoodFieldModel(double, double, double, double, unsigned long, map_t*) in likelihood_field_model.cpp.o
      nav2_util::LikelihoodFieldModelProb::LikelihoodFieldModelProb(double, double, double, double, bool, double, double, double, unsigned long, map_t*) in likelihood_field_model_prob.cpp.o
  "_pf_update_sensor", referenced from:
      nav2_util::BeamModel::sensorUpdate(_pf_t*, nav2_util::LaserData*) in beam_model.cpp.o
      nav2_util::LikelihoodFieldModel::sensorUpdate(_pf_t*, nav2_util::LaserData*) in likelihood_field_model.cpp.o
      nav2_util::LikelihoodFieldModelProb::sensorUpdate(_pf_t*, nav2_util::LaserData*) in likelihood_field_model_prob.cpp.o
  "_pf_vector_coord_add", referenced from:
      nav2_util::BeamModel::sensorFunction(nav2_util::LaserData*, _pf_sample_set_t*) in beam_model.cpp.o
      nav2_util::LikelihoodFieldModel::sensorFunction(nav2_util::LaserData*, _pf_sample_set_t*) in likelihood_field_model.cpp.o
      nav2_util::LikelihoodFieldModelProb::sensorFunction(nav2_util::LaserData*, _pf_sample_set_t*) in likelihood_field_model_prob.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libsensors_lib.dylib] Error 1
make[1]: *** [CMakeFiles/sensors_lib.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Undefined symbols for architecture x86_64:
  "_pf_ran_gaussian", referenced from:
      nav2_util::OmniMotionModel::odometryUpdate(_pf_t*, pf_vector_t const&, pf_vector_t const&) in omni_motion_model.cpp.o
      nav2_util::DifferentialMotionModel::odometryUpdate(_pf_t*, pf_vector_t const&, pf_vector_t const&) in differential_motion_model.cpp.o
  "_pf_vector_sub", referenced from:
      nav2_util::OmniMotionModel::odometryUpdate(_pf_t*, pf_vector_t const&, pf_vector_t const&) in omni_motion_model.cpp.o
      nav2_util::DifferentialMotionModel::odometryUpdate(_pf_t*, pf_vector_t const&, pf_vector_t const&) in differential_motion_model.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libmotions_lib.dylib] Error 1
make[1]: *** [CMakeFiles/motions_lib.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< nav2_util	[ Exited with code 2 ]
```

I have tried to install navigation2 in my mac. But I got a link error messages when build a nav2_util package. It seems failed to link c function in map and pf. How can I fix this?

crdelsey, 2019-04-10:
It will take me a little while to set up a build with Clang and MacOS to reproduce this, but it looks to me like there are some missing dependency statements in `nav2_util/CMakeLists.txt`

While waiting for me to reproduce, you could try adding the the line:
`target_link_libraries(sensors_lib pf_lib)` on line 62, just after the `add_library(sensor_lib ....)` statement. 

There might be more problems like this in the file. I'll take a look to try to find them, but if you find any others, please update this issue.

crdelsey, 2019-04-10:
We also need to add the line `target_link_libraries(motions_lib pf_lib)`. 

I'm unable to reproduce this with Clang 6 on Ubuntu 18.04. @routiful If you can try the changes in PR #657, I'd appreciate it. I don't currently have an OS X machine to test with.

routiful, 2019-04-11:
After I add target_link_libraries as you mentioned, the error messages are dissappeared.
https://github.com/routiful/navigation2/commit/ba30c25941752b88054faaf378b566bf970fa2b3
Thanks :)

crdelsey, 2019-04-11:
Great! Thanks for testing this.

PR #657 (2019-05-06): Adding missing dependencies in the nav2_util CMakeLists.txt

---

## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | fixes #656 |
| Primary OS tested on | OS X |
| Robotic platform tested on | NA |

---

## Description of contribution in a few bullet points

* The nav2_util CMake file is missing some `add_target_dependencies` statements. Shared libraries need to link with their dependencies. Sensors_lib has a dependency on pf_lib that was not specified. For some reason this did not cause a problem on Ubuntu 18.04 building with GCC but  [... cut]
closed 2019-04-11 by routiful