ros-navigation/navigation2 #659: Building nav2_map_server fails

Namnodorel, 2019-04-14:
- Operating System: **Ubuntu 18.10**
- Installation type: **source** _(both navigation2 and ROS2)_
- Version or commit hash: _(navigation2 downloaded by `initial_ros_setup.sh`, ROS2 latest commit on master)_

I tried building navigation2 from source, first implicitly through the Turtlebot3 building process described [here](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#pc-setup) , and after that by running `colcon build --symlink-install` in `navigation2_ws`. As far as I am aware, all packages navigation2 depends on are installed and up-to-date.

Unfortunately, I am experiencing the following error, which fails the build:

````
Starting >>> nav2_map_server
--- stderr: nav2_map_server                                                                                                                                                                         
/home/ubuntu/turtlebot3_build/navigation_isolated/navigation2_ws/src/navigation2/nav2_map_server/src/map_server.cpp: In member function ‘void nav2_map_server::MapServer::getParameters()’:
/home/ubuntu/turtlebot3_build/navigation_isolated/navigation2_ws/src/navigation2/nav2_map_server/src/map_server.cpp:78:18: error: catching polymorphic type ‘class YAML::Exception’ by value [-Werror=catch-value=]
   } catch (YAML::Exception) {
                  ^~~~~~~~~
/home/ubuntu/turtlebot3_build/navigation_isolated/navigation2_ws/src/navigation2/nav2_map_server/src/map_server.cpp:86:18: error: catching polymorphic type ‘class YAML::Exception’ by value [-Werror=catch-value=]
   } catch (YAML::Exception) {
                  ^~~~~~~~~
/home/ubuntu/turtlebot3_build/navigation_isolated/navigation2_ws/src/navigation2/nav2_map_server/src/occ_grid_loader.cpp: In constructor ‘nav2_map_server::OccGridLoader::OccGridLoader(rclcpp::Node*, YAML::Node&)’:
/home/ubuntu/turtlebot3_build/navigation_isolated/navigation2_ws/src/navigation2/nav2_map_server/src/occ_grid_loader.cpp:59:18: error: catching polymorphic type ‘class YAML::Exception’ by value [-Werror=catch-value=]
   } catch (YAML::Exception) {
                  ^~~~~~~~~
/home/ubuntu/turtlebot3_build/navigation_isolated/navigation2_ws/src/navigation2/nav2_map_server/src/occ_grid_loader.cpp:67:18: error: catching polymorphic type ‘class YAML::Exception’ by value [-Werror=catch-value=]
   } catch (YAML::Exception) {
                  ^~~~~~~~~
/home/ubuntu/turtlebot3_build/navigation_isolated/navigation2_ws/src/navigation2/nav2_map_server/src/occ_grid_loader.cpp:73:18: error: catching polymorphic type ‘class YAML::Exception’ by value [-Werror=catch-value=]
   } catch (YAML::Exception) {
                  ^~~~~~~~~
/home/ubuntu/turtlebot3_build/navigation_isolated/navigation2_ws/src/navigation2/nav2_map_server/src/occ_grid_loader.cpp:79:18: error: catching polymorphic type ‘class YAML::Exception’ by value [-Werror=catch-value=]
   } catch (YAML::Exception) {
                  ^~~~~~~~~
/home/ubuntu/turtlebot3_build/navigation_isolated/navigation2_ws/src/navigation2/nav2_map_server/src/occ_grid_loader.cpp:107:18: error: catching polymorphic type ‘class YAML::Exception’ by value [-Werror=catch-value=]
   } catch (YAML::Exception) {
                  ^~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/map_server_core.dir/build.make:76: CMakeFiles/map_server_core.dir/src/map_server.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/map_server_core.dir/build.make:63: CMakeFiles/map_server_core.dir/src/occ_grid_loader.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:179: CMakeFiles/map_server_core.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< nav2_map_server	[ Exited with code 2 ]

````

Namnodorel, 2019-04-19:
The issue is apparently connected to Ubuntu 18.10 - using a 18.04 installation resolved the issue for me.

mkhansen-intel, 2019-04-19:
@Namnodorel - Sorry we haven't been running 18.10 and haven't planned to do so, but if you create a patch that fixes this and doesn't break 18.04 we'd at least consider merging it. If using 18.04 works for you, then that's fine also.

Namnodorel, 2019-04-20:
Unfortunately, I'm only really proficient in Java and just getting started with Python, so I'm afraid fixing this is beyond my abilities right now. 18.04 will (have to) work.

crdelsey, 2019-04-22:
@Namnodorel Thanks for digging into this. It looks like a legitimate problem to me, so I'm going to keep this one open until we can fix it. I believe the fix is simply a matter of changing the catch statements to use a reference ie. ` } catch (YAML::Exception &) {`

SteveMacenski, 2019-05-11:
@crdelsey handling this. PR to follow. I'm adding referencing handling for all exceptions across the stack as well as printing those exceptions (we have the information, may as well do something with it).

crdelsey, 2019-06-21:
fixed by #871

closed 2019-06-21 by crdelsey