ros-navigation/navigation2 #660: ROS2 navigation cannot avoid obstacles while set goal from A to B.

rafaelhuang1021, 2019-04-16:
## Bug report

**Required Info:**

- Operating System:
 Ubuntu18.04
ROS2 / Master branch
#### Steps to reproduce issue
1. launch navigation2
2. set initial pose A
3. set goal B
4. set obstacle on A to B path
 
Expect result:
ROS2 navigation can avoid obstacles while set goal from A to B.
Actual result:
ROS2 navigation cannot avoid obstacles while set goal from A to B.

rafaelhuang1021, 2019-04-16:
[image]

rafaelhuang1021, 2019-04-16:
[image]

mkhansen-intel, 2019-04-18:
I don't see the obstacle showing up in the costmap layer.  How tall is the object? If you stand in front of the robot, do you show up in the costmap layer in RVIZ?

mkhansen-intel, 2019-04-18:
I see that you haven't enabled the LaserScan plugin, can you turn that on to see if laser scans are being received?

One way to check from the command line is:
```
ros2 topic hz /scan
```

rafaelhuang-intel, 2019-04-19:
hi matt, for your question:
1. If I stand in front of the robot, I'm not show up in the costmap layer in RVIZ
2. laser scan date can received after send goal from initial pose to point B

[image]

intel@intel-desktop:~/turtlebot3$ ros2 topic hz /scan
average rate: 30.294
	min: 0.030s max: 0.036s std dev: 0.00080s window: 32
average rate: 30.295
	min: 0.030s max: 0.036s std dev: 0.00058s window: 63
average rate: 30.297
	min: 0.030s max: 0.036s std dev: 0.00049s window: 94
average rate: 30.297
	min: 0.030s max: 0.036s std dev: 0.00043s window: 125
average rate: 30.300
	min: 0.030s max: 0.036s std dev: 0.00040s window: 156
average rate: 30.304
	min: 0.030s max: 0.036s std dev: 0.00039s window: 187
average rate: 30.304
	min: 0.030s max: 0.036s std dev: 0.00036s window: 218
average rate: 30.304
	min: 0.030s max: 0.036s std dev: 0.00034s window: 249
average rate: 30.304
	min: 0.030s max: 0.036s std dev: 0.00032s window: 280
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00031s window: 311
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00030s window: 342
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00028s window: 373
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00027s window: 404
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00026s window: 435
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00026s window: 466
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00025s window: 497
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00024s window: 528
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00024s window: 559
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00023s window: 590
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00023s window: 621
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00022s window: 652
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00022s window: 683
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00022s window: 714
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00021s window: 745
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00021s window: 776
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00021s window: 807
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00020s window: 838
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00020s window: 869
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00020s window: 900
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00020s window: 931
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00019s window: 962
average rate: 30.303
[... 72 lines cut ...]
	min: 0.030s max: 0.036s std dev: 0.00015s window: 2109
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00015s window: 2140
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00015s window: 2171
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00015s window: 2202
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00015s window: 2233
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00015s window: 2264
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00015s window: 2295
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00015s window: 2326
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00015s window: 2357
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00015s window: 2388
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00015s window: 2419
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00015s window: 2450
average rate: 30.303
	min: 0.030s max: 0.036s std dev: 0.00015s window: 2481

mkhansen-intel, 2019-04-19:
> 1. If I stand in front of the robot, I'm not show up in the costmap layer in RVIZ

We need more information, as again we don't see this issue where objects aren't showing up in the costmap layers.  If I stand or put and object in front of the robot (or the side or behind for that matter), we see it show up. I will post a screenshot.

mkhansen-intel, 2019-04-19:
Can you tell me what checkboxes are enabled in your RVIZ window? If you can attach your rviz config file that would be even better.

mkhansen-intel, 2019-04-19:
Here is a screenshot with the path the robot took to avoid obstacles on the map and navigate from A to B. I am running this with all the latest changes in the PRs, which we'll merge ASAP. As you can see the robot sees and avoids boxes, people and other robots. 
[image]

This is in simulation but we'll also post a video or screenshot(s) from running on the physical turtlebot3 in the lab.

mkhansen-intel, 2019-04-20:
[image]

mkhansen-intel, 2019-04-20:
Here's our lab setup. Notice when we created the map there were no obstacles
[image]

[image]

And here is the turtelbot3 navigating around an obstacle (a box) we put in the center of the map after it was created.

https://photos.app.goo.gl/8yVpVbSig9GjagAu6

LeonJung, 2019-04-23:
@mkhansen-intel 

what I'm doing also shows same results as @rafaelhuang-intel said; seems like local costmap is not being drawn on the map. 

it just pushed anything on the road away, nothing changed on its path.

so does any current PRs that have any solutions for this?

crdelsey, 2019-04-23:
@LeonJung 

We found out that the problem @rafaelhuang-intel ran into was caused by running the turtlebot 3 provided launch script with default options. ie. `ros2 launch turtlebot3_navigation2 navigation2.launch.py`

This script is using the navigation 2 parameters provided by Robitis. These parameters are a bit out of date with the latest improvements we've made in the navigation stack. Most importantly, they do not turn on obstacle detection. The latest set of parameters we've been using for Turtlebot 3 are located here: https://github.com/ros-planning/navigation2/tree/master/nav2_bringup/launch/nav2_params.yaml

You can use these latest parameters either by using the navigation 2 bringup scripts in the nav2_bringup package. See documentation here: https://github.com/ros-planning/navigation2/tree/master/nav2_bringup. Or you can use the params option provided by Robotis' launch script. ie
`ros2 launch turtlebot3_navigation2 navigation2.launch.py params:=<path to nav2_bringup/launch/nav2_params.yaml`

Please let us know if that solves your problem.

LeonJung, 2019-04-24:
@crdelsey 

Great, now it works!

mkhansen-intel, 2019-04-24:
@rafaelhuang-intel - can you verify this works for you and if so close this issue?

rafaelhuang-intel, 2019-04-25:
@mkhansen-intel @crdelsey  Now it works, thanks for all, this issue closed.

rafaelhuang-intel, 2019-04-25:
@mkhansen-intel could help to merge above command line to tutorial?

closed 2019-04-25 by rafaelhuang-intel

mkhansen-intel, 2019-04-25:
Actually the command in our nav2_bringup is already correct, if you use our nav2_bringup the nav2_params.yaml file is used by default.