ros-navigation/navigation2 #675: Robot swings left and right during the move

LeonJung, 2019-04-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- type:
  - bin
- Version or commit hash:
  - 0.1.5 (since it seems not so different with the latest)
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Expected behavior

Moving, roughly saying, straight to the goal pose

#### Actual behavior

Swings to left and right during the move_to_goal.

#### Additional information

[image]

which parameter should be optimized?

and do you have any customized dynamic reconfigure node (Best if it is QT or terminal ui version)? 
I'm also going to dive into this project... seems there is something I can contribute for.

crdelsey, 2019-04-24:
This problem was typically caused by the DWB controller not running fast enough. Try building a release version. `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release`

Newer versions seem to have better performance as well, so you may want to try the latest on the `master` branch.

> do you have any customized dynamic reconfigure node?

No. We don't have anything yet. It is something we've talked about, but haven't had time to work on yet.

> I'm also going to dive into this project... seems there is something I can contribute for.

That would be great! We're happy to have the help :-)

LeonJung, 2019-04-24:
@crdelsey 

Okay, it works. Thank you.

BTW, let me keep writing further questions in this issue.
Tell me if I should make new issue per question in this repository.

* When I give `2d goal pose` input during the `control task`, it stops then doesn't get any more goal pose inputs. Is this comes from behaviour tree ported in the nav task? and to go more deeper on this question, is this behaviour intended? ; there is a rule like navigation node won't get any other pose during path planning ordered previously`

* When I give `2d goal pose` input, sometimes the node doesn`t even start the job after showing the following statement.

```
Begin navigating from current location to (1.85, 5.79)
```
Also it doesn't get any other inputs just described in the first one.
I'm launching once again if this occurs, so basically it can be solved by own hands.
But I guess something like time synchronization stuffs was the problem.

Your comments were helpful

mkhansen-intel, 2019-04-24:
@LeonJung - if you build from our master branch and use the nav2_params.yaml file in our nav2_bringup/launch folder, you should get all the latest fixes and parameters. Please give that a try and file any new issues you observe.

crdelsey, 2019-04-24:
> When I give 2d goal pose input during the control task, it stops then doesn't get any more goal pose inputs

This is a problem. I've just recently become aware of it, but hadn't filed an issue yet. That's done now. #676. @mjeronimo is currently changing the behavior tree to use the new Action interface. It is probably best to wait until that is done before fixing this problem, because those changes will likely conflict.

> When I give 2d goal pose input, sometimes the node doesn`t even start the job after showing the following statement.

I've seen that also. I don't know the root cause yet. Can you file an issue for that?

LeonJung, 2019-04-24:
@mkhansen-intel
Yes, I brought latest navigation2 packages and the yaml file you mentioned. It works fine. Thanks.

@crdelsey 
Okay, that will be great work. I couldn't trace backgrounds through seeing the git networks; (there are so many branches), so can I ask if there was no official action interface in ros2, so temporarily the behavior tree has been used and now will be replaced?

I'm thinking whether the concept of using behavior tree system on the service nodes while another behavior tree system is running on the navigation stack is the right use or not.

mkhansen-intel, 2019-04-25:
The behavior tree is independent of the official action interface, and the behavior tree is not going to be replaced. However our "nav2_tasks" were a short-term proxy for actions and those are being replaced by actions for the Dashing release.

Per your comment above that this issue is resolved, I'm closing it.

closed 2019-04-25 by mkhansen-intel