ros-navigation/navigation2 #676: Cannot send a new goal from RVIZ until the current goal is reached or fails

crdelsey, 2019-04-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu
- Installation type:
  - either
- Version or commit hash:
  - the latest on master as of 24 Apr 2019
- DDS implementation:
  - any
- Client library (if applicable):
  - NA

#### Steps to reproduce issue
1. Bring up the nav stack and RVIZ.
2. In RVIZ, provide a nav goal using the 2d nav goal button
3. Before the nav goal is reached provide a different nav goal

#### Expected behavior
* Robot navigates to the new goal and abandons the first.

#### Actual behavior
* Robot ignores the new goal

#### Additional information

SteveMacenski, 2019-05-07:
This sounds like we need the ability to preempt the action interface. Looking through the current implementation it seems like we can only abort, cancel, or succeed.

mkhansen-intel, 2019-05-14:
Here's a link to what the simple action server policy is in ROS1, which is how the ROS1 navigation worked. We should conform to this:
http://wiki.ros.org/actionlib#SimpleActionServer_Goal_Policies

Essentially, new goals should pre-empt old ones.

SteveMacenski, 2019-05-15:
Actions in ROS2 do not have preemption available. That would mean that we could use cancel as that meets the ROS2 description, but we'd need to handle the old action returning cancel cleanly.

closed 2019-06-10 by crdelsey