ros-navigation/navigation2 #682: Questions of particle convergence and global planning

LeonJung, 2019-05-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - src
- Version or commit hash:
  - 0.1.7
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Expected behavior

- particles will fairly converge after some steps moving forward (w/ oneside direction rotation after  initial pose input)
- global planning (or perhaps local planning) will be done for each period of time, getting rid of redundant movement when the object disappears (and the patches on the costmap disappears)
  
#### Actual behavior

- particles converged after rotation movement, but spread out like a fan shape after few steps moving forward.
- once goal pose was input on the map, global planning (? the red line) path was fixed on the map even after the object on the costmap was disappeared or was newly appeared, and is never updated. 

#### Additional information

- I'm using sick lidar which has not full angular sight like turtlebot3's lidar, so I'm guessing this must be the cause, but I just want to check what's the problem. 
- I'm using 0.1.7 version package as it is,,, nothing modified without the robot radius parameter. It was expected to run like navigation ros1 package did. It is confusing me that local costmap is online, but the path is not being updated.

mkhansen-intel, 2019-05-02:
If you are using our crystal released version 0.1.7, you will not be seeing the global planner update the plan based on dynamic changes to the map. That is because that release is using the simple_navigator by default. Our latest changes in the master branch are using the bt_navigator which is behavior tree based and allows for more complex flows such as parallel planning and control. If you want those behaviors, I suggest you clone and build the repo from source.

crdelsey, 2019-05-02:
> once goal pose was input on the map, global planning (? the red line) path was fixed on the map

Parallel planning is not currently enabled. We're hoping to get it in the initial Dashing release, but I'm not 100% sure it will be in by then. In the meantime, we have a replan-on-failure in our laster build on `master` as @mkhansen-intel said.

> particles converged after rotation movement, but spread out like a fan shape after few steps moving forward

What are you testing with? A turtlebot 3? A real one or a simulated one in Gazebo? @mkhansen-intel recently found a problem with the odometry output on the real turtlebot 3, which could be causing the localization problem you're seeing.

LeonJung, 2019-05-09:
@mkhansen-intel 
Thanks. I understood what you said. 
and... I found out I can't use the latest master git since the ROS2 core suddenly upgraded by itself to dashing version (I think this is not really a perfect dashing version.. it only says that it is dashing version, but the folder name is still a crystal). What I felt was that whole sources (which includes ros2 cores) are jumped over during their revision, so that's too difficult to track what is happening. 
Seems like I should use time to track what has been changed.

@crdelsey 
Okay, thanks.
I'm not using the turtlebot3. I already checked out how's TB3 running. It seems Okay on its run. What I'm using is actually the customized that uses sick571, so it has lack of angular sight. I tried using navigation of ros1 version, and it works perfectly without particle divergence, but ros2's navigation2 still not. I hope the source master branch source solves that problem, but I have no option to verify it because of what I said to @mkhansen-intel. 

* Suggestion : Is there any reason that you are maintaining whole revision in the master git stream? I found that, in recent renewal of ros2 core development, someone who seems like a key member of its development pushed his source (which has a revised source but the other sources were too old version) into the master branch, so the pushed source completely messed up each whom takes the new master source. Afaik, maintaining source on master is very risky.

* Additional question(?) : As I'm using sick571, because of its long range, rviz2 stops working by lack of swatches. Currently, I modified the limit of swatches and looks okay now, but should we tell them (through their issue bulletin) to change  the value?

Thanks for all, everytime.

crdelsey, 2019-05-09:
> I hope the source master branch source solves that problem.

I don't think there are any fixes in master that would help with the problem you see. We might have a bug in AMCL. Can we reproduce the problem by testing with the turtlebot 3 in Gazebo, but with the laser scan limited to 270 degrees like the sick571?

> Is there any reason that you are maintaining whole revision in the master git stream?

We feel it is important for us to stay up to date with the latest changes in ROS 2. After the dashing release, we will still do development on master, but we will try to keep the dashing-devel branch up to date with all the latest fixes. That way you can just use the released dashing versions.

>  I modified the limit of swatches and looks okay now, but should we tell them (through their issue bulletin) to change the value

Yes. That sounds like a good idea. I'm sure other people will have the same problem.

LeonJung, 2019-05-09:
@crdelsey 
> I don't think there are any fixes in master that would help with the problem you see.
In recent update of ROS2 desktop debian package, dashing version stuffs seems oozed to the crystal packages, and then so many errors occurred during building even when I use v0.1.7 navigation2. 
One of the error I'm stuck is that the error of Nodeoptions not found. Do you have any idea?
 
>  We might have a bug in AMCL. Can we reproduce the problem by testing with the turtlebot 3 in Gazebo, but with the laser scan limited to 270 degrees like the sick571?
That's good idea... why couldn't I think like that. I will try after fixing the issue above... I'm trying ROS2 core reinstallation with older crystal source build.

> We feel it is important for us to stay up to date with the latest changes in ROS 2. After the dashing release, we will still do development on master, but we will try to keep the dashing-devel branch up to date with all the latest fixes. That way you can just use the released dashing versions.
Okay.

> Yes. That sounds like a good idea. I'm sure other people will have the same problem.
Then, I will tag here after issuing on rviz2 bulletin

LeonJung, 2019-05-10:
@mkhansen-intel @crdelsey 
Which ros2 core version are you using now? master branch? currently I'm suffering from master branch ros2 core build error. I think nav2 will work if ros2 core succeed building

crdelsey, 2019-05-10:
Our `master` branch builds with the ros2 master (https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos)

> One of the error I'm stuck is that the error of Nodeoptions not found

This probably means you don't have the latest ros2 core, or it didn't build and install properly. You mentioned a mix of crystal and dashing ... are you building the ros2 core from source or doing a binary install? We normally just build from source, with no Crystal binary packages sourced.

> I'm suffering from master branch ros2 core build error

I normally delete the whole ros2 workspace including `src`, `build` and `install` directories. Then I do a fresh `vcs import src < ros2.repos`. If it still doesn't build, it usually means there are new dependencies listed on the [ROS 2 build instructions page](https://index.ros.org/doc/ros2/Installation/Linux-Development-Setup/). In the meantime, while you are trying to get it to build, we have a docker image with the latest successful build of the nav stack with the latest ROS 2 stuff. You can get it by running `docker run -it rosplanning/navigation2:master bash`

LeonJung, 2019-05-13:
@crdelsey 
> This probably means you don't have the latest ros2 core, or it didn't build and install properly. You mentioned a mix of crystal and dashing ... are you building the ros2 core from source or doing a binary install? We normally just build from source, with no Crystal binary packages sourced.

Yeah. The version issue seems not so simple as I thought... As long as I used Crystal debian 4th gen, there was no problem, but after suddenly comes with the error, I had no idea. At least, I started working with the source, hopefully. I think wherever warning comment on the README is written, the best the user can follow up what you are doing.

> I normally delete the whole ros2 workspace including src, build and install directories. Then I do a fresh vcs import src < ros2.repos. If it still doesn't build, it usually means there are new [...]

Thanks for your suggestion.

I will tell if something goes okay.

LeonJung, 2019-05-15:
@crdelsey @mkhansen-intel 
Here is the result.
* navigation2 master branch / dependent pkgs mentioned in repo / ros2 core master branch (All the thing was pulled in 05/13/2019) succeeded to build. So it might say that the build issue seems completely solved
* Though nav2_bringup has been brung up, actions like 'Moving to goal pose' and even Local/Global cost map' doesn't run. Same robot was tested with 0.1.7 nav2, and it worked. Also, since we couldn't find the way to see topic messages (unfortunately, master branch ros2 core - ros2 topic echo doesn't work), we had to go another way to debug faster
* Since what I wanted to do with nav2 master branch source is to get global path planning repetition function, I've backported a part of master sources to 0.1.7
1. nav2_util(without NodeOptions)
2. nav2_tasks
3. nav2_motion_primitives
4. nav2_bt_navigator     
and the robot seems worked as it did in 0.1.7(= still no path updates).

Did I miss something in this work?

p.s. Using docker image will be the next phase.

LeonJung, 2019-05-15:
@crdelsey @mkhansen-intel 
I wrote a comment 4 hours ago, but never mind. Finally, I got my robot working with global planning update function. Thanks for all the help.
Since there were a lot of errors, misbindings, functional problems in ros2 core master branch source, I ported latest nav2 source to old crystal ros2 core with little modifications (removal of NodeOptions, that was all).
I thought that waiting for upcoming dashing stable ver is the best for me now. I'll let you know further issues and solutions(if I have), of course, during my application work over nav2 stack.

You can close this issue.

crdelsey, 2019-05-16:
> I ported latest nav2 source to old crystal ros2 core with little modifications

Ok. That's great! Would you be willing to submit a PR to update our crystal-devel branch with those changes? I think a lot of other people would like that as well.

LeonJung, 2019-05-17:
@crdelsey 

I uploaded PR, but since I can't follow up the main stream, I let that PR having nav2 master source pulled yesterday. As I mentioned earlier, I modified only the lines related with NodeOptions... just take a look.

If you want to let me upload usable one, let me know when the master branch development stream comes stable (I mean, the time that the source will not be modified frequently).

And, I think that there will be showing more errors, because ros2 core master source you are using must be very different from what I'm using.

closed 2019-07-03 by crdelsey