ros-navigation/navigation2 #691: Proposal: remove motion primatives

SteveMacenski, 2019-05-07:
We probably shouldn't be having anything except the controller send velocity commands to the base, or else we end up with a number of async or fighting issues. Having a separate server to process these commands seems a little unnecessary and adds to the complexity of the project. 

I propose that the actions of spinning, backup, stop can remain, but they should be header primatives for the controller and recoveries to use, not act as a separate server. For things like stop or backup, I think we'd be fine just removing those primatives altogether as backup should really only be called in 1 particular recovery case (probably doesnt have use else where) and stop is just sending an empty `Twist()` it might be overkill to try to separate that out.

crdelsey, 2019-05-08:
I'm inclined to agree with you, but do you have any thoughts on how to recover when the controller says it can't move? Does the controller have a special recovery params that can be set to change it's behavior, or do we just fix the controller to always report correctly whether it can move or not.

I think there are several problems with the current DWB implementation that need to be fixed before we can remove the motion primitives.
1. The obstacle footprint critic seems to say the robot is on an obstacle when it really isn't. @bpwilcox is looking into that right now.
2. We need to make DWB work with a costmap that is in the base_link frame (maybe without the static map data). This way, if AMCL jumps our position, the obstacles don't suddenly appear on the robot. I've been working on this on and off, but I've been running into problems in the DWB code where it transforms the global plan into the base_link frame.

bpwilcox, 2019-05-08:
In MoveBase, recovery behaviors were instanced as plugins, using the costmap and local planner methods to perform recoveries while checking for collisions. I'm currently working on integrating collision checking within our motion primitives, however there is a design refactor involved since the current motion primitives are independent of the costmap and controller. Part of this refactor involves grabbing the costmap and robot footprint from the world model instead of instancing the `costmap_2d_ros` in the controller, which we have proposed before. In this proposal, we could place motion primitives as servers or instance within the controllers as internal recovery behaviors.

SteveMacenski, 2019-05-08:
If `motion primatives` are actually `recovery behaviors` we should rename them as such. The name primatives by itself is asking people to call these servers as basic building blocks for other task development. 

With the server-oriented rearchitecture of nav2 I agree having the separate servers aligns with that. I suppose then if that's the way these are meant to be used for, `nav2_recoveries` would make most sense. I would be more than happy to do all those changes and PR @bpwilcox if you agree.

bpwilcox, 2019-05-08:
I agree that a rename to recovery makes sense for how we are _currently_ using them, however, I can imagine using the same behaviors as a building block for complex tasks or user commands even if not recovery-related. For instance, if I wanted to execute a spin outside of recovery, then this would make sense I think. Currently, our recovery behaviors are configured via the primitives in the behavior tree, so if anything, I might propose we create a `nav2_bt_recoveries` folder that contains some stock recovery behavior tree templates that we can drop into the bt_navigator as opposed to hard-coding the primitives into the `nav2_bt_navigator` behavior tree xmls.

mkhansen-intel, 2019-05-09:
My main goal here is to remove the hard-coding of the recovery flow and behaviors. The 'motion primitives' idea was to create generalized actions that could be executed either as part of the recovery flow of a behavior tree, or separately if desired. The latter part is semi-questionable since we haven't really ID a use case where a motion primitive is needed other than recovery. So I'm fine with renaming or replacing them, as long as we leave the ability to create them as actions executed from the BT engine. I also do agree that they need to check for collisions. 

At least to me, this is one of the benefits of moving away from move_base hard-coding behavior flow to the BT in the first place, is to add this extensibility / configurability.

SteveMacenski, 2019-05-09:
Ok, I'll take this on

mhpanah, 2019-05-09:
@mkhansen-intel Motion primitives are also being used in Auto Localization in addition to recovery behaviors. 

@SteveMacenski I agree that we shouldn't have anything except the controller to send velocity commands, but the motion primitives are suppose to be a simple controller to either spin the robot or actuate the robot forward/backward based on angle or distance.  Also, we should not actuate the robot before checking the direction of motion for obstacles. 

Currently the motion primitives have lots of redundant code for simple velocity commands that they are sending.  I think we should just have a single simple controller for this purpose.

orduno, 2019-05-10:
On renaming the package: If `nav2_motion_primitives` are *currently* exclusively being used for recovery, I'm ok with renaming to something like `nav2_recovery_motions` (not sure if we could use "behavior" here).

On moving the primitives to the controller: Currently, dwb is a path-based controller, and the approach suggested requires to extend the functionality of dwb to implement simple motions like forward, spin, etc. If that's the consensus, I would be inclined to define a `nav2_controller` package which internally specifies a follow_path controller (dwb, timed elastic bands) and primitives/recovery controller. That would simplify the architecture, avoid code duplication and address the problem of having a second server sending velocity commands.

If the decision is to keep the primitives/recoveries as a separate package and avoid having multiple nodes publishing to \cmd_vel, then having a `robot` class would be useful.

SteveMacenski, 2019-05-10:
Since the recoveries are handled through the behavior tree, I think we can allow velocity command to come out of a _recovery behavior_ server like they are now, just renamed to make it clear that they are not primatives for use, but default recoveries exposed by a server. If in recovery mode, the navigator shouldn't be letting the controller do "stuff" so there shouldn't be async artifacts.

As I have thought about how I'd personally implement a system like this, I'm a believer that all commands should funnel through a single "thing" (lets call it a controller for this case). Now that doesn't mean the controller needs itself to contain these base actions, but rather that these actions call the publishers within the controller such that the controller at the end of the day is the one sending the commands to the base. This lets us have access in the controller server to the state of recovery in case of asynchronous interactions. (i.e. we expose a topic that it funnels to the base, and when that is called we have some muxing lock-out of the controller while recovery actions are being taken).

SteveMacenski, 2019-05-15:
PR #705 covers, but the spirit of the original ticket was to remove, we decided to rename, so closing

closed 2019-05-15 by SteveMacenski