ros-navigation/navigation2 #692: Proposal: remove robot class

SteveMacenski, 2019-05-07:
In going through the code the robot class doesn't seem to add much (for the time being). It's only used in 3 places to do the following:

Getting pose -- which should probably be done through TF anyhow

Sending velocity to base -- which should only be done by the controller, that can just be 1 line config for topic. 

Its an additional package that adds a certain level of complexity without much existing benefit. I propose removing the robot class entirely, and revisiting this concept when other elements of the system are stable. I like the idea of a centralized class where we can define the motion model, certain high level parameters, and description information, however it doesnt seem to be on the priority list of things to work on and the existing state of the project makes it hard to read and understand where this is potentially going. 

We've had alot of good ideas and thought but this one in particular wasn't fully thought out and I think its worth removing and restarting when we have better defined requirements and there's someone willing to put extended time into pulling *all* the robot specific elements into that class. 

Lets discuss below. If we agree, I can do this.

crdelsey, 2019-05-08:
After you pointed out that the current architecture results in too many connections in the rqt_graph, I think we're all on board with this.

My thinking was that the subscribers that are in nav2_robot right now could become utility classes in nav2_util, kind of like is done with the odom_subscriber in DWB (https://github.com/ros-planning/navigation2/blob/master/nav2_dwb_controller/nav_2d_utils/include/nav_2d_utils/odom_subscriber.hpp)

I'm not sure if this approach works for the "get pose from transform" problem though.

mkhansen-intel, 2019-05-08:
I agree, since the robot class is not really fully designed, in it's current state it is not useful and should be removed. If at a future date we have a need and someone takes the time to do a design first, we can revisit.

SteveMacenski, 2019-05-09:
Ok, I'll take this one

mhpanah, 2019-05-09:
I also agree.

closed 2019-07-19 by SteveMacenski