ros-navigation/navigation2 #707: Minimal example for using the repo functionalities

bergercookie, 2019-05-13:
Quoting the README overview section:

> The ROS 2 Navigation System is the control system that enables a robot to autonomously reach a goal state, such as a specific position and orientation relative to a specific map. Given a current pose, a map, and a goal, such as a destination pose, the navigation system generates a plan to reach the goal, and outputs commands to autonomously drive the robot, respecting any safety constraints and avoiding obstacles encountered along the way.

Is there a minimum working example of this behavior? If not can you give me / point me to some instructions on how to use it, assuming almost minimum knowledge of the ROS1 navigation stack (i.e., costmaps, global/local planners)?

There's a relevant question on ros.answers but I thought that the minimum working example is specific to this repo: https://answers.ros.org/question/323051/current-status-of-the-navigation-stack-tools-in-ros2-where-is-move_base/

mkhansen-intel, 2019-05-13:
Our [nav2_bringup README](https://github.com/ros-planning/navigation2/tree/master/nav2_bringup) has the directions for bringing up the stack on a Turtlebot3 in Gazebo. The process is essentially the same on a real robot also.

To answer your question about move_base. We have split move_base up into it's base components in ROS2 and are using behavior trees to drive the components. This provides customizability and extensibility of the stack behavior. However, if you use our nav2_bringup launch files, you don't need to worry about that to get started using the stack.

bergercookie, 2019-05-13:
I see, thanks for clarifying.
I've started experimenting with the nav2_bringup package..

It look a bit convoluted for a first-time example to be honest. And also demands that the user compiles packages like `turtlebot` and `turtlebot_simulations` which in turn bring `cartographer` and `amcl` (which in turn demand ceres to be installed -> suitesparse etc.

I think it would come very handy if a super basic example was to be added, e.g., upload a sample world file with no obstacles for gazebo, a sample diff drive robot, and have `navigation2` compute a sample path..

Thanks for the help.
P.S. Any other resources / examples on how to use this?

mkhansen-intel, 2019-05-13:
Sorry if it feels convoluted to you, we've had quite a few people use it without dependency issues. We basically worked with Robotis to enable the Turtlebot3 with Nav2, which we feel is actually better than a basic robot, but we'd be open to adding further examples. If you want to contribute one via a PR, we'd like that.

bergercookie, 2019-05-15:
I finally got it to work (sort of)

As a suggestion the description can point to https://github.com/ROBOTIS-GIT/turtlebot3/tree/master/turtlebot3_navigation/maps instead of asking the user to record and save a map of its own.

The main problem IMHO and at this stage is that the rviz2 version being distributed with `crystal` on Ubuntu 18.04 looks fragile. For start  it doesn't subscribe to the /map topic that `amcl` publishes unless you explicitly restart rviz. And then *sometimes* you have to restart it again so that you can send a navigation goal to the robot.

Finally whatever the navigation goal,the robot starts moving in an endless circle.

bergercookie, 2019-05-15:
Regarding the last error I mentioned, I'm getting the following output when setting the 2D pose estimate from rviz:

```
[WARN] [amcl]: Failed to transform initial pose in time (Lookup would require extrapolation into the future.  Requested time 1557933040.78233 but the latest data is at time 139.01300, when looking up transform from frame [odom] to frame [base_footprint])
[INFO] [amcl]: Setting pose (138.926000): 0.489 -0.414 -0.015
[INFO] [amcl]: AmclNode publishing pose
```

Then when issuing a navigation goal, the robot starts moving but the `amcl` estimate isn't getting updated at all even though the robot is moving.

`ros2 run tf2_ros tf2_echo map base_link

```
- Rotation: in Quaternion [0.000, 0.000, -0.032, 0.999]
At time 629.767000000
- Translation: [0.952, -0.377, 0.010]
- Rotation: in Quaternion [0.000, 0.000, -0.032, 0.999]
At time 630.767000000
- Translation: [0.952, -0.377, 0.010]
- Rotation: in Quaternion [0.000, 0.000, -0.032, 0.999]
At time 631.767000000
- Translation: [0.952, -0.377, 0.010]
- Rotation: in Quaternion [0.000, 0.000, -0.032, 0.999]
At time 632.767000000
- Translation: [0.952, -0.377, 0.010]
- Rotation: in Quaternion [0.000, 0.000, -0.032, 0.999]
At time 633.768000000
- Translation: [0.952, -0.377, 0.010]
- Rotation: in Quaternion [0.000, 0.000, -0.032, 0.999]
```

bergercookie, 2019-05-15:
I've cloned and compiled the navigation stack from source. Doing a bit more diggging, it seems that (as stated before) the amcl filter doesn't update the map->odom transform, unless when I explicilty initialise it with the `2D Pose Estimation` button from rviz2.

```
...
      tf2::impl::Converter<true, false>::convert(odom_to_map.pose, latest_tf_);
      latest_tf_valid_ = true;
      if (tf_broadcast_ == true) {
        // We want to send a transform that is good up until a
        // tolerance time so that odom can be used
        auto stamp = tf2_ros::fromMsg(laser_scan->header.stamp);
        tf2::TimePoint transform_expiration = stamp + transform_tolerance_;
        geometry_msgs::msg::TransformStamped tmp_tf_stamped;
        tmp_tf_stamped.header.frame_id = global_frame_id_;
        tmp_tf_stamped.header.stamp = tf2_ros::toMsg(transform_expiration);
        tmp_tf_stamped.child_frame_id = odom_frame_id_;
        tf2::impl::Converter<false, true>::convert(latest_tf_.inverse(), tmp_tf_stamped.transform);
        std::cout << "[amcl_node.cpp:901] DEBUGGING STRING - NEVER FIRES ==> " << 18 << std::endl;
        this->tfb_->sendTransform(tmp_tf_stamped);
        sent_first_transform_ = true;
      }
    } else {
      RCLCPP_ERROR(get_logger(), "No pose!");
    }
  } else if (latest_tf_valid_) {
    if (tf_broadcast_ == true) {
      // Nothing changed, so we'll just republish the last transform, to keep
      // everybody happy.
      tf2::TimePoint transform_expiration = tf2_ros::fromMsg(laser_scan->header.stamp) +
        transform_tolerance_;
      geometry_msgs::msg::TransformStamped tmp_tf_stamped;
      tmp_tf_stamped.header.frame_id = global_frame_id_;
      tmp_tf_stamped.header.stamp = tf2_ros::toMsg(transform_expiration);
      tmp_tf_stamped.child_frame_id = odom_frame_id_;
      tf2::impl::Converter<false, true>::convert(latest_tf_.inverse(), tmp_tf_stamped.transform);
      std::cout << "[amcl_node.cpp:920] DEBUGGING STRING - ALWAYS FIRES ==> " << 19 << std::endl;
      this->tfb_->sendTransform(tmp_tf_stamped);
    }
```

I'm running off the `crystal` branch as I've installed crystal via the ubuntu 18 binary packages
Any suggestions?

crdelsey, 2019-05-16:
> the amcl filter doesn't update the map->odom transform, unless when I explicilty initialise it with the 2D Pose Estimation button from rviz2.

This was by design. It didn't seem to make sense for AMCL to publish a transform if it has no idea where in the world it is. It would almost certainly be wrong, leading to problems in the rest of the stack. Do you have a use case where you'd want to start navigating without setting an initial pose?

> I'm running off the crystal branch as I've installed crystal via the ubuntu 18 binary packages

Our crystal branch is pretty out of date right now. We've been focussed on Dashing since it comes out in about 2 weeks. However, there seems to be a number of people trying to use navigation with Crystal, so I'll investigate backporting so it works better.

bergercookie, 2019-05-16:
>This was by design. It didn't seem to make sense for AMCL to publish a transform if it has no idea where in the world it is. It would almost certainly be wrong, leading to problems in the rest of the stack. Do you have a use case where you'd want to start navigating without setting an initial pose?

No, you 're right. The problem in my case is that the navfn_planner node needs to know the current pose while navigating the environment. So I set the pose explicitly in rviz, amcl gets that and posts a single updated pose. After that though, when I set the navigation goal, the planner produces a valid path and sends velocity commands to the robot so that it follows that path. After that though it needs feedback of where the robot is so it keeps querying for the current pose of the robot via the `/amcl_pose` topic. The problem is that there is never a new pose sent by the amcl node to that topic! Does this make sense?

I imagine this is because of incorrect initialization of the amcl filter for some reason.

P.S. This is my understanding of the navigation stack. Let me know if I got something wrong. :-)

crdelsey, 2019-05-16:
AMCL should keep publishing updated /amcl_pose messages. It does this based on robot movement. If the robot moves a certain amount it publishes a new pose. If it isn't doing that, either your robot is not moving very far or the /odom messages are not being received correctly.

SteveMacenski, 2019-07-19:
@crdelsey should this ticket be closed? It looks like it brings up questions that over tickets represent

crdelsey, 2019-07-19:
Let's keep this open for a little while longer. I'm going to try to write a bunch of docs over the next week. This is one of the topics I need to address.

mkhansen-intel, 2019-09-18:
@bergercookie - I believe we can close this now. Please open a new issue if you're experiencing any more issues. Hopefully things are working well for you.

closed 2019-09-18 by mkhansen-intel

bergercookie, 2019-09-18:
I'll check and reopen if I find something.
Thanks Matt.