ros-navigation/navigation2 #712: Implement on_error in lifecycle nodes

mjeronimo, 2019-05-14:
## Feature request

#### Feature description
Currently, the lifecycle nodes do not do any error handling/recovery in the on_error callback. We should implement an error strategy consistently across the lifecycle nodes. For example, the nodes might notify the lifecycle manager upon an unrecoverable error and the manager could attempt to restart the node (deactivate, cleanup, configure, activate).

mkhansen-intel, 2019-06-07:
Adding some additional info.

It seems we have some exceptions that cause things to crash during bringup without going to the on_error state at all:
```

9: [world_model-5] [INFO] [world_model]: Activating
9: [world_model-5] [INFO] [global_costmap.global_costmap]: Activating
9: [world_model-5] [INFO] [global_costmap.global_costmap]: Checking transform
9: [world_model-5] terminate called after throwing an instance of 'eprosima::fastcdr::exception::NotEnoughMemoryException'
9: [world_model-5]   what():  Not enough memory in the buffer stream
9: [ERROR] [world_model-5]: process has died [pid 8893, exit code -6, cmd '/opt/nav2_ws/install/nav2_world_model/lib/nav2_world_model/world_model __params:=/opt/nav2_ws/src/navigation2/nav2_system_tests/params/nav2_params.yaml'].
```
And other cases where it is trapped and does go to the on_error() state.

I added code to dwb to catch and print ERROR messages in this commit: c5b916f6c6f4939ab032109f32161e044b254724

That catches them and prints the error, then tries to return a FAILURE to the lifecycle manager. I don't think the lifecycle manager handles it yet, but that should be the goal I think.

BTW - to make it fail, re-add the ObstacleFootprintCritic in DWB, you should see this error:
```
9: [dwb_controller-7] [INFO] [dwb_controller]: Using critic "ObstacleFootprint" (dwb_critics::ObstacleFootprintCritic)
9: [dwb_controller-7] [ERROR] [dwb_controller]: Couldn't initialize critic plugin!
9: [dwb_controller-7] [ERROR] [dwb_controller]: Couldn't load critics! Caught exception: std::exception
```

crdelsey, 2019-07-08:
From @mkhansen-intel : Maybe we can inform lifecycle manager when we get to on-error so that lifecycle can take some action - maybe go back to unconfigured state.

SteveMacenski, 2019-10-15:
@crdelsey from our discussion last week, it sounds like our direction is to move away from the manager and towards something else. Is there something we want to do with `on_error`? I dont think we necessarily *have* to do anything and that's totally valid. 

The best option I see is to have error plugins that lets a user inject whatever they want to have happen in an error case (notify something, restart, start a counter, etc) if we're not going to provide a manager

crdelsey, 2019-10-21:
>  it sounds like our direction is to move away from the manager and towards something else.

I think we want to migrate the ideas of lifecycle manager into the launch tool. 

Besides making on_error plugins, I think a reasonable thing to do would be to report the error state back to the launch tool, whereupon the failed node is killed and restarted, or something. For a truly robust system, we probably need both.

In the short term though, I think having the on_error method just report out the error and terminate the program is reasonable. I thought that was already implemented though.

SteveMacenski, 2019-10-21:
I dont have a strong preference here. I really don't even know what I would do with an `on_error` function, so maybe someone more creative than me can give me some intuition. 

Assuming its been launched, I would expect there to be some param for `respawn` like in ROS1 to bring the process back up after a failure, so I'm not sure what the `on_error` should do if relaunching is on the launch provider. Cleanup assets? Log about it? Try to take down other related services?

crdelsey, 2019-10-22:
I just checked, and every lifecycle node reports the error and exits. Closing this issue.

closed 2019-10-22 by crdelsey